prusa3d/Prusa-Firmware-Buddy #1294: [BUG]<After update to 4.3 lcd jump everywhere

Af5211, 2021-02-08:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI+]

**Printer firmware version**-  4.2.1 to 4.3

**Original or Custom firmware** Original

**Optional upgrades** - [Filament Runout Sensor + super pinda]

**USB drive or USB/Octoprint**
 USB flash drive

**Describe the bug**
  After update the lcd display go everywhere 

**How to reproduce**
  Please describe steps to reproduce the behavior.

**Expected behavior**
  After update from 4.2.1 to 4.3 no controle on lcd 

https://user-images.githubusercontent.com/78602880/107217745-e93b5c00-6a0e-11eb-9c12-bc61b96b99ed.MOV

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

JohnnyDeer, 2021-02-10:
@Af5211 thank you for reporting. Could you please check if LCD wires are assembled properly? Also i would like ask you for upload photo of label with printer ID, thank you.

Af5211, 2021-02-10:
After downgraded to 4.2.1 the lcd works fine.

JohnnyDeer, 2021-02-15:
@Af5211 We would like to know which series on LCD you have, please upload photo of label with printer ID, thank you.

JohnnyDeer, 2021-03-11:
@Af5211 Did the replaced LCD sent from our Support solve this problem?

Af5211, 2021-03-11:
It solve the problem. Thank you for the support.
Greetings Antoon Friesen

closed 2021-03-15 by JohnnyDeer