prusa3d/Prusa-Firmware-Buddy #1297: [BUG] since 4.3.0 occasionally bluescreen after filament change

lost-carrier, 2021-02-09:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]
MINI

**Printer firmware version**-  [e.g. 4.0.5, ...]
4.3.0

**Original or Custom firmware** [Original/Custom]
Original

**Optional upgrades** - [e.g. Filament Runout Sensor]
Yep, Filament Sensor

**USB drive or USB/Octoprint**
Usually Octoprint, but happens when on the built in UI...

**Describe the bug**
Since installing 4.3.0 it occasionally happend now a couple of times for me.
I want to change filament. After heating up and stuff, the printer says "Ramming..." and ejects the old filament.
When done with this, the Mini throws some kind of bluescreen along with some loud beeping - see attached image.
Reseting the printer, and selecting Filament -> loading filament works fine.

**How to reproduce**
As said: happens only occasionally - not sure how exactly is needed to reproduce, but...
- Filament -> Change Filament
- Wait until "Ramming"
- Happens when (or shortly before) Ramming is done...

**Expected behavior**
Dialog with selection for next filament to load appearing.

**G-code**

**Crash dump file**

**Video**
[image]

ea4ep, 2021-02-10:
I have a same problem.

JohnnyDeer, 2021-02-10:
Hi @lost-carrier @ea4ep thank you for reporting. Unfortunately I'm unable to reproduce it. Please try to observe more information around the bug, ideally find scenario how to reproduce step by step. Thank you for your help.

ea4ep, 2021-02-10:
> Hi @lost-carrier @ea4ep thank you for reporting. Unfortunately I'm unable to reproduce it. Please try to observe more information around the bug, ideally find scenario how to reproduce step by step. Thank you for your help.

Is when make a filament discharge.

I have video and photos.
[image]

ea4ep, 2021-02-10:
Is impossible upload video by mobile.
This is YouTube link.

https://youtu.be/vUYr0SWC6Gg

Thanks.

lost-carrier, 2021-02-10:
@ea4ep What is your firmware version? Can't properly read it, because of the sticker. Is it also 4.3.0? And actually I have a slightly different (though still similar) text in the error message...

ea4ep, 2021-02-10:
> @ea4ep What is your firmware version? Can't properly read it, because of the sticker. Is it also 4.3.0? And actually I have a slightly different (though still similar) text in the error message...
> 
> ...but: yes - happens when unloading the filament.

Hi, my version are the last on GitHub, I compile my version for BMG extruder, only make a step change and Z max value to 183
The txt version said 4.3.1 in last day, but in some days the version in txt file are 4.3.0

Regards.

JohnnyDeer, 2021-02-15:
Hi @lost-carrier @ea4ep, could i ask you to try to reproduce it on 4.3.1-RC1? Thank you. 
I'm still unable to reproduce this bug.

ea4ep, 2021-02-15:
> Hi @lost-carrier @ea4ep, could i ask you to try to reproduce it on 4.3.1-RC1? Thank you.
> I'm still unable to reproduce this bug.

Yes, where are located a RC1 ver?
Link for download?.
Thanks.

JohnnyDeer, 2021-02-15:
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.1-RC

ea4ep, 2021-02-17:
> https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.1-RC

Hello, I have tested the RC1 for two days and at the moment there is no trace of the blue screen.  It works well.

JohnnyDeer, 2021-02-22:
@ea4ep thank you for update. I´m closing this issue as solved :)

closed 2021-02-22 by JohnnyDeer

lost-carrier, 2021-02-22:
...just for completeness sake: installed the 4.3.1 final mid of last week - did not have the issue since then. Thanks for fixing!