prusa3d/Prusa-Firmware-Buddy #1305: [BUG]<Downgrade 4.3.1 rc to 4.2.1 not possible.

Af5211, 2021-02-13:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version**-  [e.g. 4.3.1 rc]

**Original or Custom firmware** [Original

**Optional upgrades** - [Filament Runout Sensor + superPINDA

**USB drive or USB/Octoprint**
 USB flash drive 

**Describe the bug**
  Downgrade from 4.3.1 RC to 4.3.0 or 4.2.1 impossible 

**How to reproduce**
  After reset is not possible for downgrade

**Expected behavior**
  Make it possible to downgrade

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

murk-sy, 2021-02-13:
Just to be sure, did you remove all 4.3.1 firmware file from the drive?

Af5211, 2021-02-13:
Yes I have 
Only the 4.2.1 file is on the usb-stick

JohnnyDeer, 2021-02-15:
Hi @Af5211 how exactly is it not possible? Are you using reset button to start flashing? Do you have allowed FW Updates in settings?

Af5211, 2021-02-15:
it is solved, turn on the printer, after booting hold the button and reset button as well, and downgrade.

JohnnyDeer, 2021-02-15:
@Af5211 thank you for updating issue, I´m closing.

closed 2021-02-15 by JohnnyDeer