prusa3d/Prusa-Firmware-Buddy #1309: [BUG]after several minutes into a print the REST values disappear

0xygen8, 2021-02-17:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI
**Printer firmware version**-  4.3.1
**Original or Custom firmware** Original
**Optional upgrades** -  No Filament Runout Sensor
**USB drive or USB/Octoprint** USB
**Describe the bug**
when the printer is connected to the network and powered the REST values are readable, when a print is started the values disappear (Unavailable)
**How to reproduce**
start a print, and wait for a couple of minutes (5+)

JohnnyDeer, 2021-02-18:
Hi @0xygen8, thank you for reproting. The values disappear on website or on printer LCD?

0xygen8, 2021-02-18:
I found the reason, while the printer is idle the REST values reply direct, when the printer is printing I need to raise the timeout for the values to 30 seconds to keep them available in my automation. This is new for the 4.3.x ,4.2.0 did not have this issue.

0xygen8, 2021-02-18:
tied to issue #1285 ?

closed 2024-06-17 by github-actions[bot]