prusa3d/Prusa-Firmware-Buddy #1315: No "user-friendly" change for the print fan.

Erik262, 2021-02-20:
**Printer type** - [MINI]

**Printer firmware version**-  [4.3.1-2759]

**Original or Custom firmware** [Original]

**Optional upgrades** - [nope]

**USB drive or USB/Octoprint**
USB drive

**Describe the bug**
I have a problem with an announced feature, which didn't release.

_Setting the RPM manually

The RPM of the fan is adjusted using a PWM factor, which can be changed in LCD Menu -> Settings->Temperature->Print Fan. In the current release, due to a historical reason (MK3) the firmware is displaying the direct PWM value (0-255), **in the future release, this will be changed to the more user-friendly range of 0-100 %**. The speed of the print fan can be also set using a G-code command M106 Sxxx, where the xxx is a value in the range 1-255._

In which version are we talking about?? I thought it would be 4.3.1 ?

**How to reproduce**
Just try to change the print fan while you are printing :)

**Expected behavior**
Changing the print fan in a percent way instead in a range of 0 - 255.

**G-code**
No g-code

**Crash dump file**
nope no crash

**Video**
Nope, my prusa is very shy ;P

Prusa-Support, 2022-05-17:
Hello! Sorry for the very late response to this thread. I changed the label, will bring this to our devs attention so that it can be evaluated and implemented - I can confirm it that it's currently still with the 0-255 range (4.3.4). 

Alessandro Pantaleo
Prusa Research

Erik262, 2022-12-23:
Hello there, the "issue/bug" still exist in the latest FW 4.4.0. The range between 0-255 for the print fan doesn't give any meaningful explanation for non-computer people. Especially for younger for you do active marketing.

BTW: In the settings we are able to display the print fans RPM. 
My suggestion:
While running the "Self-Test", why not saving the highest reached RPM for calculating the print fans percentage? or by just doing some simple math? -> PrintFan ≈ (x*255)/100

closed 2024-06-17 by github-actions[bot]