prusa3d/Prusa-Firmware-Buddy #1317: [BUG] PC blend turns into PVB after firmware update

colingg, 2021-02-24:
I'd like to report it, although its a very uncommon bug.

**Printer type** - MINI

**Printer firmware version**-  4.2.1 -> 4.3.1

**Original or Custom firmware** Original

**Optional upgrades** - N/A

**USB drive or USB/Octoprint** - N/A

**Describe the bug**
When you load in PC blend on 4.2.1 upgrade to 4.3.1 it shows PVB after upgrade, forcing you to reload filament and wasting PC blend 💸💸💸

**How to reproduce**
1) flash to 4.2.1
2) load in PC blend
3) flash to 4.3.1
4) PC turned into PVB

**Expected behavior**
Keep PC blend loaded as setting, instead of showing PVB

**G-code**
unrelated

**Crash dump file**
N/a

**Video**
N/a

JohnnyDeer, 2021-03-18:
Hi @colingg, I can confirm this bug. Thank you for reporting.

JohnnyDeer, 2021-05-17:
We decided not to fix this bug because it is unlikely to happen to more amount users. Our priority should be set to more common and important bugs. Thank you for understanding.

closed 2021-05-17 by JohnnyDeer