prusa3d/Prusa-Firmware-Buddy #1318: [FEATURE REQUEST] Rename Custom Sheet Profiles

Gromit21, 2021-02-24:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
Printer type - MINI+
Optional upgrades - SuperPinda

Describe the expected functionality

The ability to rename the custom sheet profiles in the Settings menu.
This will help to match the profile to the multiple custom steel sheet being used.

Gordontgopher, 2021-02-26:
This would be especially useful as the satin sheet is now available for the mini.

Dev01-D, 2021-03-17:
Super helpful as i have all 3 sheets and we could also have aditional such as Glass etc.

ulab, 2021-04-20:
I was quite confused, because I am able to rename the sheets on my MK3S, but not on the Mini+. 

And now I have to use my satin one as "Custom1" sheet which doesn't do it justice.

Kalimero13, 2021-04-24:
Change the configuration using a file on a USB flash drive, similar to an upgrade firmware.
For example, the contents of a text file (config.ini):

[sheet]
1=smooth PLA
2=satin  PLA
3=textured PLA
4=smooth PETG
5=satin PETG
6=textured PETG

It is possible to change other constants in EEPROM with this configuration file.
For example, setting the steps for the extruder when replacing the extruder:

[step]
E=415

ulab, 2021-04-26:
Do you have a link to any kind of documentation on this? I can't find anything related to the Mini and a config.ini file?

Kalimero13, 2021-04-26:
This is my idea, the link leads to my head.

TerTerro, 2022-01-11:
Would second this. It's annoying to not have option to change names and like the option of USB config file just for sheet names for example.

DiomedesDominguez, 2023-05-08:
I'd love to know 2 things:
1) How many sheets configurations we can create if we could use the whole or at most the 90% of the EEPROM space for this task
2) Can we select the sheet profile through gcode or any special code from the prusaslicer?

Let's say that we can have up to 12 sheets instead of 6. This way we can have not only for PLA and PETG, but also for ASA/ABS and TPU

Prusa-Support, 2023-06-20:
Thank you all for bringing this to our attention.

We may consider an option to rename the sheet profiles in the future but at the moment there are only 8 pre-sets available: 2x smooth sheet profiles + 2x textured sheet profiles + 4x custom sheet profiles.
These profiles are pre-set and can't be renamed but please notice, except for the naming, there are no differences between them. The user may save any value on any profile, regardless of what the sheet is actually called.

This menu was modified recently in the past by introducing satin sheet profiles, but I seem to understand these were accidentally removed. This may change again in the near future.

Michele Moramarco
Prusa Research

TeslaCT, 2023-08-06:
I still think this should happen. I would also like to see if we could link the sheet profile to a nozzle. I use my .6 nozzle a lot. I can and will use some of the other profiles for when I use different nozzles. but without being able to rename the profiles it could be dangerous to use the profiles.  my .6 CHT nozzle is very different than the standard brass. so if I confuse the profiles I could easily crash the nozzle

Kyeld, 2023-10-09:
The ability to rename the profiles would be a nice quality of life update for the Buddy firmware, its a bit of a let-down that the current generation of printers lack this feature.

SnowB0und14, 2024-02-19:
It also appears to be missing from the MK3.5 upgrade as well.  I wonder if they could at least add the satin sheets to factory list.

big-fudge, 2024-04-01:
For an outsider this seems like a pretty quick change. The idea is around for over 3 years now and still might dust away in some backlog. Is there a priority problem with this or is it just not planned and won't be in the feature list for future updates alt all? Could someone from @Prusa-Support please clarify? It really would help A LOT to have any option to rename those sheets. Especially with the presence of different nozzle diameters and materials to print on different sheets.

m-kozlowski, 2024-04-17:
This FR really needs some love.
Just finished upgrading four mk3s to mk3.5 and of all quality of life features lost in the process, this is the one I miss most.

danopernis, 2024-04-17:
This issue is on our radar. Suggested solution using INI file is valid, but feels a bit hack-ish. We are debating implementing a virtual keyboard for our printers, which would unblock this and many other feature requests.

m-kozlowski, 2024-04-17:
Hackish? Maybe. But having a choice between manually renaming 8 labels using touchscreen keyboard or uploading and applying preconfigured file, I'll take the ini way, hands down. Especially having multiple printers to configure.

big-fudge, 2024-04-17:
What he said.

maciii, 2024-04-23:
I will also like it on MK3.5.

jsrumsey, 2024-05-08:
I would also like the renaming of steel sheet profiles in MK3.5, like we used to have.

CZDanol, 2024-05-16:
Hello, we're working on this feature.
Internal ticket: BFW-4898

big-fudge, 2024-05-16:
Thx for the heads up, but that doesn’t cover a custom naming for the sheets. The offset doesn’t do anything for me. Nor the sheet name. I still can’t do anything with that in terms of used nozzle diameters or sheet/nozzle combinations. 

Would be useless as it currently is with just the offset attached to it.

CZDanol, 2024-05-16:
@big-fudge I don't understand?

big-fudge, 2024-05-16:
Bc it’s the essence of the linked ticket?

CZDanol, 2024-05-16:
Yes, and?

murk-sy, 2024-08-26:
Implemented in [6.1.2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.1.2)

CZDanol, 2024-08-26:
Indeed :) Closing the issue.

closed 2024-08-26 by CZDanol