prusa3d/Prusa-Firmware-Buddy #1320: [BUG] Mesh Bed Leveling Silently fails during first layer calibration

anthr76, 2021-02-25:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version**-  4.3.1

**Original or Custom firmware** Original

**USB drive or USB/Octoprint**
USB Drive

**Describe the bug**
During first level calibration if mesh bed leveling probing fails the first level calibration still continues to start

**How to reproduce**
Have a serious XZ axis skew, and do mesh bed leveling.

**Expected behavior**
First layer calibration should exit and fail like it would for a normal print.

**Video**
  Please attach a video. It usually helps to solve the problem.
https://photos.app.goo.gl/R4VBho1fdHu1WsYM8

Sadly in this video I don't have the begging where the probe hites all the points of the bed but it silently failed after the fourth probe and started printing. Resulting in a etched bed.

I'm one day old to 3D printing so sorry if my terminology isn't perfectly correct. I can supply more information. Upon setup of my Prusa Mini I etched my print bed due to a big skew in my XZ leveling. Shouldn't first level calibration fail like it would on first prints?

JohnnyDeer, 2021-03-01:
Hi @anthr76, this seems like broken MINDA cable, please contact our Tech Support for help. Chat in right down corner https://shop.prusa3d.com/en/

anthr76, 2021-03-01:
> Hi @anthr76, this seems like broken MINDA cable, please contact our Tech Support for help. Chat in right down corner https://shop.prusa3d.com/en/

Thanks for your prompt reply @JohnnyDeer I have indeed escalated this with customer support and they have resoled the issue. My rods are not fully fastened and there was a substantial XZ skew. The right side was much lower then the left. Though the bug I'm reporting here (unless it's expected) the probing stage should fail on a first layer calibration like it would for a regular print. My MINDA probe/cable was just fine.

First layer calibration was letting me continue with the skew while a normal print would not proceed after the fourth probe. I would get a bed leveling error. Shouldn't the first layer calibration do that as well?

JohnnyDeer, 2021-03-03:
@anthr76 Yes, I can confirm that it should behave the same in both situations. Thank you for reporting.

Dario503, 2021-05-03:
Prusa Mini+ (circa 11/2020), v4.3.1 release firmware.  XZ axis was skewed, approximately 3mm end to end on x-axis.  Manually invoking Mesh bed leveling stops at 4th location and then the display goes back to the menu with no error, no warning that the leveling did not complete.  If a print is sent (via Octoprint for example) it will perform the incomplete bed leveling and proceed with the print which ends poorly  due to the XZ axis skew.  It should not print if the mesh bed leveling fails or at least give a warning.

wouterokkerse, 2022-09-21:
this is still an issue in 4.3.4 firmware. mesh levelling stops but print still starts on first layer calibration and when a print starts.

Prusa-Support, 2022-09-27:
Thank you for your input.

When the PINDA doesn't work well, the anomaly would be immediately noticeable, but I must agree that an error message would make the debugging easier.

I'm afraid that - because the geometry of the MINI is extremely simplified - the printer is expected to work within a substantially large range of calibration values without throwing an error. It could be hard to narrow down such range reasonably enough, to prevent fake errors but I'd let the developers think about it.
Meantime, the issue will stay open.

Given that the described behavior, as of now, is the expected one, I would tell this is a known issue, and not really a bug, but a request of adding a feature - such as an error message.

Michele Moramarco
Prusa Research

wouterokkerse, 2022-09-28:
> Thank you for your input.
>
> When the PINDA doesn't work well, the anomaly would be immediately noticeable, but I must agree that an error message would make the debugging easier.
>
> I'm afraid that - [...]

hi,

i noticed today that one of our mini's on firmware 4.3.3 did give an error of not completing mesh levelling while the one on 4.3.4 did not. is it possible that such a feature was removed in a firmware update.

closed 2024-06-17 by github-actions[bot]