prusa3d/Prusa-Firmware-Buddy #1321: [BUG] Clock Drift?

reedy, 2021-02-25:
**Printer type** MINI

**Printer firmware version** 4.3.1

**Original or Custom firmware** Original

**Optional upgrades** Filament Runout Sensor

**USB drive or USB/Octoprint**
USB

**Describe the bug**
It seems even over the period of 90 minutes or so, the Prusa Mini firmware clock seems to drift quite a bit.

**How to reproduce**
Start up the printer, it seemingly gets the time via NTP, after running a print job of around 90 minutes, the clock is 2-4 minutes behind the actual time

**Expected behavior**
The clock on the device to not drift by multiple minutes over a period as short as 90 minutes.

**G-code**
N/A

**Crash dump file**
N/A

**Video**
N/A

reedy, 2021-02-25:
I just had to cancel a print... Within 30 minutes, the clock had seemingly drifted ~2 minutes

reedy, 2021-02-25:
[image]

[image]

reedy, 2021-02-26:
Comparatively... After a ~5H print... It was only still about 2 minutes out...

I guess if the display displayed the current time (at the top, like a phone/tablet?) the drift wouldn't be so annoying. I don't have to get another device to compare the difference etc

reedy, 2021-02-26:
#1322 file to display the time on the device, to make it more obvious to see what's going on..

#648 would also help "fix" this in a different way too

tbenz9, 2021-11-19:
I too have noticed the clock drift is quite severe.  Seems like syncing against an NTP server on a frequent basis (say once/minute) would fix this minor annoyance.

Eldenroot, 2022-11-23:
Displaying time in the top part of the display would be best

Prusa-Support, 2022-11-30:
Thank you all for reporting the issue.
The issue will remain open to let our developers look into it.

Also, mind there are several similar issues already opened.
For example, please discuss matters that concern the User Interface (how the time should be displayed on the top) on the separate issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1322.

Michele Moramarco
Prusa Research

Prusa-Support, 2022-12-16:
The new firmware [4.4.1-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1-RC1) improves the estimated end time thanks to the adoption of the RTC (Real Time Clock), and possibly solve this issue.

This issue is probably directly related to issue #1067 and solved together with it.

Could this issue be closed?

Michele Moramarco
Prusa Research

reedy, 2023-01-30:
It doesn't seem any better on 4.4.1...

[image][image]

reedy, 2023-01-31:
Another print today. 3-4 minute drift over a less than 2 hour print..

Prusa-Support, 2023-02-14:
At the current state of the art, the print time is not extremely precise.
For peace of mind, I've gathered some information that I'll try to explain my best.

The print time is calculated by Prusa Slicer based on the theoretical machine model. In the reality, the commands execution time is influenced by hardware tolerances and conditions, and may slightly vary from printer to printer, from print to print. 

Because the calculated estimation can't be absolutely precise, the G-codes include M73 codes for every estimated elapsed minute of print time (not necessarily exactly equal to 60 seconds in the real life). This code will update the remaining time (minutes) and percentage on the display. In case of user intervention of any kind, the time will be offset until the next M73 code occurs during the print.

Also, the printer clock time accuracy depends on the server. The clock time refresh doesn't occur that frequently, maybe once or a few times every hour (unconfirmed information) which is considered acceptable for the purpose of the print estimate end - which unfortunately won't be extremely accurate anyway.

I can speculate that the print time can vary up to 10% (or so) from the calculated estimation.

These time drifts seem to be acceptable. This doesn't seem the case of a bug.

Michele Moramarco
Prusa Research

reedy, 2023-02-14:
It's one thing the calculation of the time to finish the print being wrong, it's another the clock drifting so much over a short period.

If it's right at startup, or when it's done the NTP call, it is not the case of "accuracy depends on the server."

You claim the RTC implementation fixed the issue, but it clearly hasn't.

Prusa-Support, 2023-03-02:
I'm not totally sure if we are not looking at a bug.
However, our developers may work on higher clock precision and more accurate print time estimates.

For now, please keep in mind the clock and estimated end time may not be very precise due to the following reasons. 
- The RTC is not battery-backed.
- The server time refresh occurs roughly once every hour.
- The remaining print time is refreshed every time that code `M73` occurs (approximately every minute of active printing). This means that slower-than-expected movements and print interruptions can't be accounted for, but within ≈1m the g-code will trigger an update.
- The print end time is the product of combining remaining time and server time, and both may slightly vary before the next refresh (≈1h for the clock / ≈1m for the print end estimate).

Michele Moramarco
Prusa Research

nvtkaszpir, 2023-05-12:
On my Prusa MINI+ it looks like the time is taken from NTP servers but device itself shows time skew around 2min after 20min of syncing time.
I guess there are two issues here:
- RTC clock use is incorrect so time is causing to lag behind the actual value
- device not getting time synced from NTP servers frequently enough (though 1h should be really sufficient, could be more frequent for servers taken from DHCP options)

Prusa-Support, 2023-06-10:
Thanks for highlighting that.
The time may not be very precise as of now but should adequately autocorrect at a certain rate.
It looks like a mixed hardware and firmware limitation, but we can work on the latter.

Michele Moramarco
Prusa Research

nvtkaszpir, 2023-12-16:
@Prusa-Support any progress on this?

Prusa-Support, 2024-01-12:
The clock and the time estimation work fairly well for the purpose of 3D printing, despite not being extremely precise.
There is surely room for improvement but sadly improvements in these regards are not in the current plans yet.
During the last year, our developers worked tirelessly and released more than a dozen firmware _stable_ releases in this repository, bringing major improvements to old and new printers.

This issue will remain open as a suggestion for further development plans.

Michele Moramarco
Prusa Research

reedy, 2024-05-21:
Still valid.

Prusa-Support, 2024-12-03:
In the final firmware release 6.2.0 the NTP server should be changed to a local one that checks the time every ~ 20 min.
This should help however, as already said in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1321#issuecomment-1889808560, I wouldn't expect the printer clock to be very precise but reliable enough for time referencing on simple 3D printer.

Michele Moramarco
Prusa Research

Prusa-Support, 2025-02-15:
The final release of FW 6.2.0 which includes server change and ~20 min refresh rate, is public.
Is the issue still relevant?

Michele Moramarco
Prusa Research

nvtkaszpir, 2025-02-15:
I'm going to update firmware this week and will check it. Please stay tuned.

closed 2025-04-24 by github-actions[bot]