prusa3d/Prusa-Firmware-Buddy #1322: [FEATURE REQUEST] Display current time on screen

reedy, 2021-02-26:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** MINI
**Optional upgrades** Filament Runout Sensor

**Is your feature request related to an existing problem? Please describe.**
Partially related to #1321

**Describe the expected functionality**
Display the current time (as per the internal MINI clock) on the display; like would be done on a phone.

It would make issues like #1321 less annoying, as you can see the relative time to the print finishing, based on the internal clock. And you also don't need to go find another device to check the time.

See also: #648 would be another way of not needing this.

Noxmiles, 2021-03-11:
Good Idea, but it should have Internet for that. So, in the best case together with the WiFi Upgrade :)

reedy, 2021-03-11:
I have mind plugged in via Ethernet ;)

mrog, 2021-06-07:
In addition to adding the current time, I'd like to see the remaining time on the screen, too.  The current behavior when there's an internet connection is to replace the amount of time remaining (e.g. "Remaining time 1h 52m") with the time of completion (e.g. "Print will end Today at 13:14").

It would be helpful to see the current time, the end time, and the amount of time remaining all at once.

Eldenroot, 2022-11-23:
With 4.4.0_RC is possible to use wifi module (I do)... why no feedback from Prusa team?!

reedy, 2024-05-21:
Still valid.

CZDanol, 2024-05-24:
I'd say that the main thing preventing us from displaying clock on the mini is that we don't have enough space in the header. It doesn't look like that at the first glance, but we have to keep the space for all the icons, including connect transfer progress.

CZDanol, 2024-05-24:
Maybe it could be placed somewhere in the print progress screen?

Prusa-Support, 2024-05-24:
Good point, that would require a major UI redesign.
Maybe an optional footer would do? 🙂

Michele Moramarco
Prusa Research

CZDanol, 2024-05-24:
That would be implementable :)