prusa3d/Prusa-Firmware-Buddy #1324: [BUG] Z axis crash in self test / xyz test after upgrade to 4.3.1

szabotibor, 2021-03-02:
**Printer type** - MINI+

**Printer firmware version**
-  4.3.1

**Original or Custom firmware** Original

**Describe the bug**
After FW upgrade to v 4.3.1 i'm getting crash on Z axis when running self test / xyz test. All the other test are ok. Z axis moves either all the way up and stops and fails or in some cases it went all the way up and back down and failed. When running Self test all other test are OK except the Z axis.

I already tried to realign pinda height, checked the cables, downgraded FW to 4.2.1 -> Everything works as expected when using FW 4.2.1

**How to reproduce**
  - Upgrade to v 4.3.1
  - Run xyz test or self test

**Video**
  https://youtu.be/h9frvyBVcWM

JohnnyDeer, 2021-03-03:
Hi @szabotibor, this looks to me like HW issue. Could you please try look to sensor info what status is MINDA, right after failed test?

szabotibor, 2021-03-03:
Hi @JohnnyDeer, i downgraded the FW to 4.2.1 just to be sure, so i can't check the sensor status, anyway i have SuperPINDA installed. I did a quick check in the source code and found that there is Z axis max set to 191 in the rewritten self test which was introduced in v 4.3.0 as you can see in the video - mine stops at 194.16mm so i believe it has something to do with this. Also i found a thread in Prusa mini users group in facebook with the same issue: https://www.facebook.com/groups/prusaminiusers/permalink/3985352254861425/

Turro75, 2021-03-03:
with nozzle tip touching the bed which is the lactual ength of the exposed z rod? 194.16mm? if yes in some way the hotend assembly must be reviewed to reduce that dimension down to 191 (I did it on mine by adjusting the distance heatblock/heatsink and pinda position).
If the actual lenght is less than 191 the z motor is losing steps so reassembly the z axis as some constraints are preventing a full movement of the leadscrew. (try loosing a bit the leadscrew nut)

pkg2000, 2021-03-11:
@szabotibor Have you installed any mods, like the Bondtech extruder?

szabotibor, 2021-03-11:
I've checked FB thread i linked previously for new info, based on a comment there (https://www.facebook.com/groups/prusaminiusers/permalink/3985352254861425/?comment_id=3990457364350914&__cft__[0]=AZXA46vDmm0iRIk0mh-Kghn2RA65-yGe9QhiSEsg9AURmqEh85iNnK0dXq-VqDlNXBXLle8C3fS5wBzTBG9gRRZy-aYqPmCi648s2zhrtWbeSCW5WWlWsIEpscesqk3Kzgr_rbHxUJ8YEKKkwUyDeR1O&__tn__=R]-R) this is an issue with bondtech heatbreak and the new self test. And yes i have bondtech heatbreak and extruder installed.

[image]

JohnnyDeer, 2021-03-18:
We will work on this in one of next releases. Thanks for reporting and help.

stepikovo, 2021-03-18:
A temporary solution can be using a ~4mm spacer on top of the Bondtech extruder to compensate for the missing "bump" on the original extruder which acts as a stop, which prevents the axis to travel too far.

[image]

JohnnyDeer, 2021-09-08:
We released new RC with hidden menu where you can adjust the length of Z-axis based on your need so this issue should no longer appears. Please try it out and let know how it works :) 
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.3-RC

closed 2021-11-08 by JohnnyDeer