prusa3d/Prusa-Firmware-Buddy #1333: [FEATURE REQUEST]Silent Stepper mode

Dev01-D, 2021-03-17:
Would be keen to see the Silent mode option enabled in the firmware. Coming from other marlin based printers i was a little disapointed to hear the noise from the default settings on the mini+

I believe that this feature is already available on the Mk3s and can't be too much work to enable stealthchop / spreadcycle in the firmware.

I would flash my own compiled marlin but don't want to have to physically break the board to enable this.

JohnnyDeer, 2021-03-18:
Hi @Dev01-D, this issue is duplicate of #1236, I´m closing this one.

closed 2021-03-18 by JohnnyDeer

cedricwalter, 2021-04-23:
my prusa mini make a LOT of noise (not mounted by myself), I put it an enclosure but it is still noisy, so much that i'm considering getting rid of it (was having problem with neighbours), picture will follow and video. If it can be solved with a firmware then please consider doing something