prusa3d/Prusa-Firmware-Buddy #1340: Z Axis Crashing

Alexander-T-Moss, 2021-03-20:
When you unload/load filament it raises the Z axis up and this will sometimes crash into the top of the Z axis ( if done a couple times or Z axis is raised after print ) is there any way to turn on a Z collision check like it does in the self test to stop it crashing or home axes before raise so it will be 30 ish mm of the bed ?

riodoro1, 2021-03-29:
It’s been almost [a year](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/414)(since this particular issue)  and we are still waiting on _any_ kind of crash detection.

Some days I feel like this project has been on a back burner for a while. No progress on big features and no word from the manufacturer. Thanks Prusa for engaging the community so well.

mrog, 2021-08-02:
I'm seeing this issue on a Prusa Mini+ with firmware 4.3.1.  You can also crash the Z axis using the "Move Axis" controls in the settings without first running auto home.  The firmware doesn't seem to detect the crash in either case.

zoltan-l, 2022-01-15:
@Alexander-T-Moss 
Can you please double check?  It can happen after you switch on a printer and the had is in higher position and you run the load/unload more times. When you switch off the printer it does not store the actual position to the eeprom memory due to the reason that you can move the head manually, so after switch on again the stored information becomes useless. This could be changed by running a homing process after each switch on even in case you do not want to print.  This is a trade off.
Before any print the printer runs the homing process so it knows where the head is. So if the crash happens also after the print without switch of, let us know please. A short video would be great.

hertzi, 2022-03-03:
@zoltan-l 
I was, like @Alexander-T-Moss, thinking that behavior is a bug. Your explanation is right though, it only happens when the printer was just switched on and the z axis was still in a high position. 'auto home' in calibration fixes the problem, but not 'move axis' in settings , since the first value is usually 0.
I see two issues here. 
1. 'move axis' in z direction should also move into negative direction until it finds true 0 (just like it 'auto home')
2. 'load/unload/change filament' should auto home first, since each of this tasks will do the 'parking' first, even if your run unload and load after each other and the unload run the parking already and the z value reads '20' now (meaning you have a higher chance to run into the crash if you do 'unload' and 'load' instead of 'change' which runs the 'parking' process just once).

ra-kal, 2023-01-31:
This bug is really anoying and potentionally destructive to the motor/Z-axis.

I've seen a one liner PR #2184 to always enable Z-axis crash detection, so hope it will finally get into the FW. Even though this one is already 6 months old. Another option would be to enable crash detection during loading or disable Z-axis rising.

The fix seems to be pretty simple.

platzh, 2023-01-31:
> Hello! Thank you for your idea, sorry that this aspect is annoying. It is
> not expected to cause damage to Z-axis or its motor, or any component. I'll
> report to devs to evaluate a possible modification, though nothing we can
> promise so far - you can prevent it by auto-homing the printer before the
> fil. change procedure.
>
> Alessandro Pantaleo
> Prusa Research

While that is true, it’s taking time.

Why just don’t move the Z-axis at all?

Most of the time, it is that I after a print change the filament.
Depending on the model I print this is the reason the printhead is quite
high. If I print the next model it will autohome either way.

closed 2024-06-17 by github-actions[bot]