prusa3d/Prusa-Firmware-Buddy #1341: Changing PID Values in Configuration_A3ides_2209_MINI.h not possible

felixhesshaw, 2021-03-20:
I tried to modify my PID values because I have a bondtech extruder and edited the values in the config file 'PID Values in Configuration_A3ides_2209_MINI.h'. 
Compiling is not problem but when I read the PID values out with pronterface I get the old values and not the new ones.

Why are the values from the config file not applied to the firmware?

Best regards

poblopuablo, 2021-05-04:
Not sure if this will work but try this:

Re-flash firmware, then m501 (to load the setting to eeprom) then m503 to check.

JohnnyDeer, 2021-05-06:
Hi @felixhesshaw, unfortunately this won't work because we are using our own temperatures control. Thank you for understanding.

matthewlloyd, 2021-06-28:
I've added this feature to my unofficial Llama Mini firmware, which you can download here: https://github.com/matthewlloyd/Llama-Mini-Firmware/#readme

SnickCH, 2022-09-01:
I disagree. I made the test with different hotends and with a new 50W heater cartridge. It makes a difference if you tune PID (with octoprint) or just let the standard as it is. The difference: with no tuning it needs a way to long time until the heat is set. At the beginning I have a long period where the heat is overshooting for 10-15degrees. To handle this, I need 4-10x skirt until the heat is down to the point it should be. The same goes for filament load / unload. 

I kindly ask to integrate this because it makes live much easier

closed 2024-06-17 by github-actions[bot]