prusa3d/Prusa-Firmware-Buddy #1345: [BUG] Confirmation for color change filament unload

Chips83, 2021-03-26:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version**-  [4.3.1]

**Original or Custom firmware** [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint** - [USB]

**Describe the bug**
  If i add a color change in PrusaSlicer the MINI is doing an unload of the filament and then asks me if the unload was successful.

**How to reproduce**
 Add a color change in PrusaSlicer

**Expected behavior**
  If the color change GCODE is reached, my MINI should ask if i want to do the unload now and waiting for my confirmation. I want this behavior because i am not at the printer all the time. And if the unload is done unattended then the filament is not wound up properly.

fortinmike, 2021-03-28:
I've had the exact same gripe with my MINI when doing color layered prints.

Maybe there is a reason for the immediate unload? I know it's not recommended to keep filament loaded for too long while the hot end is heating and not printing, as this can sometimes cause clogs.

So maybe the behaviour could be changed to:

1. Reach the color change. Keep heating the hot end (at 170 degrees to prevent oozing).
2. Keep filament loaded and wait a few minutes in case the user notices the printer has stopped and can perform the color change (this would cover my case as I'd notice the printer beeping within 5 minutes most of the time). **The user would need to confirm before filament is unloaded.**
3. If the user does not do anything and the timeout elapses, **cool down the hot end to prevent clogs, but keep filament loaded**.
4. When the user comes back, he's prompted to perform the color change. Hot end heats back up and the normal filament change procedure runs as usual (starting with filament unload).
5. Print resumes printing with the new color.

JohnnyDeer, 2021-03-29:
Hi @Chips83 thank you for your suggestion but this is not a bug, this is enhancement.

adynis, 2021-03-30:
I vote for this ! :)

salexan2001, 2022-04-23:
I also would like to use this enhancement.

beejiujitsu, 2022-07-12:
> I've have the exact same gripe with my MINI when doing color layered prints.
> 
> Maybe there is a reason for the immediate unload? I know it's not recommended to keep filament loaded for too long while the hot end is heating and not printing, as this can sometimes cause clogs.

The Prusa Mini will automatically cool down the hotend after 30 mins of inactivity, so this is not an issue.

fortinmike, 2022-07-12:
I have come to the conclusion that this should stay as-is. When I commented, I was quite new to 3D printing and thought that behaviour was unexpected. However, I now think it's better to keep it exactly like it is now:

- Unloading the filament early is better to avoid partial or full clogs due to the filament staying heated in the hot end for too long. Clogs are much more bothersome than a not-perfectly-wound spool.
- The current behaviour saves time when you do attend to your printer, because you don't have to wait for the hot end to heat up again to perform a filament change. Now _that_ can waste some time. With the current behaviour, you just insert the new filament and press a button.
- The spool winding not being perfect is a non-issue. As long as it's not tangled (which it can't be just due to an unload), there is no problem with there being a bit of slack until you can attend to your printer and perform the filament change.
- If the winding slack bothers you, an option is to use an auto-rewinding spool holder. There are multiple of those on Printables and Thingiverse.

I'd personally vote for this issue being closed.

beejiujitsu, 2022-07-19:
> The spool winding not being perfect is a non-issue. As long as it's not tangled (which it can't be just due to an unload),

This is not necessarily true. Anecdotally, yesterday I used a spool that had been previously unloaded unattended; it did bind on itself at some point.

> The current behaviour saves time when you do attend to your printer, because you don't have to wait for the hot end to heat up again to perform a filament change. Now that can waste some time. With the current behaviour, you just insert the new filament and press a button.

Heat up time is negligible, less than a minute with PLA temperatures.

> If the winding slack bothers you, an option is to use an auto-rewinding spool holder. There are multiple of those on Printables and Thingiverse.

You're presenting a workaround, not a solution, to the problem; I would prefer the described solution.

Prusa-Support, 2022-07-28:
Thank you for your feedback and suggestions.

I believe some suggestions in this thread could be input for further improvements, but please notice there is no tangling hazard as long as the filament tip stays inserted.
For the filament on the spool to tangle, the filament tip must be free, otherwise it won't be possible to be moved under a coil of wound filament and cause a tangle.

For spool holders opened on one side (like the MK3 series stock one), two slack coils could fall on the side and cross each other. No biggies - even in this case, no tangles, but a portion of filament could get bent while manipulating and pulling the filament by the end.
In these cases, freeing a few coils and carefully re-winding is always recommended - probably re-winding much less than one meter would be all.

However, a lowered temperature and a timer could be ideas to further improve the user experience in certain ways. Maybe **a shorter unloading distance** would be enough to prevent the filament from melting and clogging, at the same time avoiding "too much" slack filament on the spool.

Feel free to close the issue if you feel this issue is a minor concern. The issue can be reopened if the community's interest in this issue will rise. Or feel free to keep it open if you'd like to push for a specific improvement to be possibly implemented sometime soon.

Michele Moramarco
Prusa Research

fortinmike, 2022-07-30:
@beejiujitsu Fair points, but I don't think a tangle is possible with a holder like the one that comes with the MINI, as @Prusa-Support mentioned. At least it never happened to me, after hundreds of unloads. I was suggesting the auto-rewinding spool as a solution if you want to keep a perfectly wound spool, if that's important to you, not to prevent tangles. As for the heat up time, it can add up. In essence, if it's relevant to change the current default, I wouldn't be ready to commit to the exact behaviour I originally described. It would probably require some more thought. The suggestions by @Prusa-Support would be interesting to consider.

beejiujitsu, 2022-08-01:
> @beejiujitsu Fair points, but I don't think a tangle is possible with a holder like the one that comes with the MINI, as @Prusa-Support mentioned. At least it never happened to me, after hundreds of [...]

You're not the original issue author, which I support.

fortinmike, 2022-08-08:
> You're not the original issue author, which I support.

I don't think being the OP has anything to do with the validity of a point, but you're entitled to your own opinion. I just think it's important that every point be considered if the default behaviour is to be changed. I'm looking forward to reading further discussion on the subject.

beejiujitsu, 2022-08-08:
> I don't think being the OP has anything to do with the validity of a point, but you're entitled to your own opinion. I just think it's important that every point be considered if the default [...]

That issue can be solved by making it a configurable option within the printer.

I doubt it'll go anywhere; Prusa seems to have abandoned any new development for the Prusa Mini in favor of the new Prusa XL.

fortinmike, 2022-08-08:
> That issue can be solved by making it a configurable option within the printer.

This could be an option, yes.

> I doubt it'll go anywhere; Prusa seems to have abandoned any new development for the Prusa Mini in favor of the new Prusa XL.

A bit off topic but it's true that there hasn't been much on that front until recently. Take a look at the just-released 4.4.0 beta firmware's release notes though: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-beta2

Looks like they were stuck on the Wi-Fi update for a while (rewriting lots of things to accommodate it) but it's coming to an end and they still managed to add a couple of really nice features and fixes in there too, such as uploading files to the printer directly, PrusaLink not being read-only, etc. Good stuff and encouraging for the future of the MINI!

beejiujitsu, 2022-08-08:
That’s encouraging.

Sent via Superhuman iOS ***@***.***>

Prusa-Support, 2022-08-18:
Our development team is continuously growing, but often development plans prove themselves to be much more demanding than the expected.
The recent MINI firmware long release notes and the Prusa XL updates on the blog are only some examples of ambitious projects keeping our developers busy. There is much more going on, among projects of various nature (Prusa Slicer, Printables.com, Prusa Academy, Prusa Enclosure, Prusa Pro AFS, maintenance legacy products, and more...) and other hidden projects which may not "take off".

Thank you all for your patience and trust.

Michele Moramarco
Prusa Research

beejiujitsu, 2022-10-11:
I don't use the holder that comes with the Mini, so your statement doesn't
apply to me.

Sent via Superhuman ***@***.***>

Prusa-Support, 2022-10-15:
That's totally fine. The spool holder is not such a crucial part of the printer.
It is important that the spool holder and its location won't cause friction over the unwinding and feeding of the filament.

A (partial) tangle between entire coil loops is very unlikely to happen, except for the use of a very long unloading distance and a multi-filament setup. This could be the case with the MMU2S, in fact it comes with a filaments buffer.
A tangled tip of the filament may occur only if the end of a filament is left free on a totally unloaded spool.

Michele Moramarco
Prusa Research

closed 2024-06-17 by github-actions[bot]