prusa3d/Prusa-Firmware-Buddy #1347: Firmware update not possible

pmapeso, 2021-03-27:
I have tried to update to the newest firmware (4.3.1), but it will not mount any USB using the knob push method or the standard of it auto-detecting firmware. Both drives I have tried are in FAT32 format. One is 16GB and the other is 6GB and neither have worked not matter what I try. I get Error #2 when I push the knob on reset for flashing older/any firmware and it will not do anything when I turn on FW update -Always or -On restart and restart/reset the printer.

JohnnyDeer, 2021-03-30:
Hi @pmapeso, please try to use more USB sticks if some will work try to flash 3.10.0-RC1 this version should help read form all USB sticks. If none of USB won´t be able to work right please contact our Tech Support. Thank you

SHAXperiment, 2021-04-01:
We have exactly the same problem. Funny thing: it worked with the firmware 4.3.1 and we printed a lot of things. Then we put a usb into it and a blue screen came up. After that no usb stick worked. Printing over USB cable or octoprint is working. But we can't upgrade the firmware.

JohnnyDeer, 2021-08-23:
As I wrote above 4.3.2 solves this problem. If none of USB won´t be able to work right please contact our Tech Support.
I´m closing this issue.

closed 2021-08-23 by JohnnyDeer