prusa3d/Prusa-Firmware-Buddy #1351: [BUG] Can't connect to serial [4.3.2-Beta1]

SM-26, 2021-04-01:
**Printer type**: Prusa Mini

**Printer firmware version**:  4.3.2 - Beta 1

**Original or Custom firmware**: _Original_

**Optional upgrades** -  Filament Runout Sensor & Spinda

**USB drive or USB/Octoprint**
Octoprint

**Describe the bug**
After flashing the latest Beta, serial isn't working any more.
When I try to connect, I get this
[image]

**How to reproduce**
1. update to ver 4.3.2-Beta1
2. (try to) connect to the printer via OctoPrint.

**Expected behavior**
Before the update, everything worked.
I only did the fw update, nothing else has changed.
I was expecting to be able to connect.

**Crash dump file**
[Dump.Bin](https://drive.google.com/file/d/185-IbF3mzUgtkbi95CUdpZ8phk9ScZqM/view?usp=sharing)

**Octoprint log**
```log
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: /dev/ttyACM0@115200, /dev/ttyACM0@250000, /dev/ttyACM0@230400, /dev/ttyACM0@57600, /dev/ttyACM0@38400, /dev/ttyACM0@19200, /dev/ttyACM0@9600
Trying port /dev/ttyACM0, baudrate 115200
Connecting to port /dev/ttyACM0, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0xa5d36350, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3831
Please see https://faq.octoprint.org/serialerror for possible reasons of this.
Changing monitoring state from "Detecting serial connection" to "Offline (Error: SerialException: 'read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3831)"
Connection closed, closing down monitor
```

**Video**
https://photos.app.goo.gl/PZYuANKqYvT4qejP6  
Sorry that the video is in Hebrew, But I think that you can still understand what is going on.

xthedakmanx, 2021-04-01:
Same error here when connecting using octoprint

[image]

gimpfenlord, 2021-04-02:
can confirm this,
downgrading to 4.3.1 solved it

SM-26, 2021-04-02:
> can confirm this,
> downgrading to 4.3.1 solved it

How do you downgrade?
I don't know how to force the printer to flash the old fw file.

gimpfenlord, 2021-04-02:
> > can confirm this,
> > downgrading to 4.3.1 solved it
> 
> How do you downgrade?
> I don't know how to force the printer to flash the old fw file.

https://help.prusa3d.com/en/guide/how-to-update-firmware-mini-mini_128421#128582
Starts @ Step 6 Downloading an old firmware

wdgeek, 2021-04-06:
I can also confirm exactly the same problem with the 4.3.2-Beta1
After the upgrade, when I try to connect via Octoprint I get the error.
I downgraded back to 4.3.1 and problem disappeared.

JohnnyDeer, 2021-04-07:
We are aware of this bug and will fix it in one of the next releases. Thanks all for reporting.

Crowlord, 2021-05-01:
While you are at it fix https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/739 as well. Its beyond the joke now.

JohnnyDeer, 2021-05-13:
This issue will be fixed in upcoming FW version 4.3.2.

JohnnyDeer, 2021-05-25:
New FW version with fix of this bug is out. Please try it out ad let know if this issue can be closed :)
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.2-RC1

wdgeek, 2021-05-25:
Just tried a test print with the 4.3.2-RC1 and I can confirm that I no longer get a bluescreen. I also noticed that there was a fix to the run out sensor and Octoprint, so I also tested that fix and it worked perfectly.!!!! Finally I don't have to worry about ruining a print when the filament runs out when printing with Octoprint. When my filament is running low I would try to remember to print from the USB drive, but I seem to always forget and I would end up with a ruined print. So happy they fixed this. THANKS.

closed 2021-05-26 by JohnnyDeer