prusa3d/Prusa-Firmware-Buddy #1352: [BUG] Mesh bed leveling does not detect difference between superPINDA and motor sensor-less end stop

markkamp, 2021-04-03:
**Printer type** - MINI+

**Printer firmware version**-  4.3.1

**Original or Custom firmware** Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
  Both

**Describe the bug**
  When the printer starts mesh bed leveling, it does not notice that the 'end stop' comes from a motor sensor-less end stop detection on the Z-axis, instead of the superPINDA sensor. This creates weird offsets in mesh leveling without the user even noticing leveling went wrong (on some of the mesh points).
  In my case the Z axis had too much friction so it randomly triggered a motor sensor-less end stop detection. Impossible to calibrate the first layer. (As the mesh bed leveling randomly corrected fake heights)

**How to reproduce**
  Start Mesh Bed Leveling. Hold Z-axis with your hands before the superPINDA sensor detects the bed. When looking at the terminal output in OctoPrint you will see a line like: Recv: echo:endstops hit:  Z:2.40

**Expected behavior**
  Give a warning that a motor sensor-less end stop was triggered on a mesh point, instead of detection with the superPINDA sensor.

**G-code**
  G28
  G29

**Crash dump file**
  not applicable

**Video**
  Here you can see an adjustment of the Z axis. 0,750mm seems right. Retry with the same settings: head suddenly plunges into the bed. 0,400mm suddenly seems the right value. Created by false mesh data. Bed visualizer in Octoprint is also a good indicator: never gave the same result.
  https://youtu.be/qTu0jSrtjgw

JohnnyDeer, 2021-04-07:
Hi @markkamp, sensorless endstop on Z should not allow MBL to be done. When i try to reproduce this i got MBL fail.
Could you please try to move superPINDA few millimeters down?

markkamp, 2021-04-09:
Hi @JohnnyDeer,

That's strange! In the 16+hours of trouble shooting I never saw an error. Did you invoke a motor sensor-less end stop on just one or two of the mesh points?
In my case: the difference was only 1 or max 2mm between the MBL runs. Maybe these values just fell in to range, and you did a more 'extreme' test example?

(This is where my knowledge stops) but does the MBL first determine the height of the bed in the first measurement? After which it does 12 measure points to measure the bed level mesh?
Could be the first measurement was the cause. I had some measurements where everything was suddenly raised 2mm. And could be that the MBL is fine with this because every point was within a small range of each other. (Just guessing here)

Moving the superPINDA only moves the sensor point, which is I'm confident to say not the problem at hand. In a 3+ hours support call with Prusa, al the basics were handled (sensor height, xz skew, etc.). 
I could definitely feel the resistance in the Z axis when I decoupled the Z axis motor and moved it by hand. After rebuilding the Z axis the problems went away without even re-adjusting the superPINDA sensor.

CarlosGS, 2022-06-27:
Oh! This also happens in the latest master branch. The stallguard endstops remain enabled for X and Y during the travels for the mesh. Any friction can cause the X or Y origins to be reset to zero during these travels, but the printer continues probing and doesn't detect there is a problem. And this yields a bad mesh bed calibration :boom:

:arrow_right: **You can replicate this problem by softly bumping the X or Y axes during bed leveling, while they are moving between probing points. The printer detects a collision and incorrectly resets that axis to 0, offsetting the entire grid.**

Unfortunately it silently fails and just resumes probing. A patch solution is to increase stall torque by setting 50 to [X and Y sensitivity](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/2ffb3c9bd0603c14dccfd95c039fd84f5f34e9ba/src/common/trinamic.cpp#L54-L55). But this could still sometimes produce an incorrect bed level.

The robust solution would be to detect these crashes (instead of silently ignoring) and notify the user that a collision has occured during bed leveling. Then it could tell users "aborting: either there is too much friction or a real collision has happened".

Prusa-Support, 2022-07-13:
Thank you for your feedback.

About the main post, I can't really tell if the issue was caused by friction on the Z movement, but touching the lead screw may not help (the Z-axis could bend).
About the friction on the Z-axis, which disappeared after a rebuild, it was likely the [trapezoidal nut](https://help.prusa3d.com/guide/3-x-axis-extruder-assembly_199519#200484) being too tight or skewed.
I can't really tell what is the actual cause of the issue is if it can't be reproduced. 
Please notice, because the printer is lightweight and the Z-axis is bound to one only lead screw at one only end, even a gentle touch may interfere with the calibration values or produce artifacts mid-layer.

If the problem does not persist, I would recommend closing the issue.

.
About the last comment, maybe this deserves to be posted and studied as a separate issue.
From my experience, the X or Y axis could produce jerk movements during the calibration for several reasons.

@CarlosGS I'm not sure if this is related to the phenomenon that you observed but a video could help (possibly on a new separate issue). The phenomenon should be analyzed further, and then a modification of the firmware could be considered to make the firmware recognize a collision. I'm not sure it will ever be possible although - the crash detection is disabled because the MINI runs in 'stealth mode' only.

Michele Moramarco
Prusa Research

CarlosGS, 2022-07-13:
Hi Michele! You are right, the main post is about the Z axis, which shouldn't have the sensorless endstop enabled. I'll try to catch it on video and move my observation to https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1131 Thanks! :grin:

markkamp, 2022-07-14:
@Prusa-Support My printer is running fine for a year now after I fixed that skewed trapezoidal nut.

But I do think @CarlosGS and I do experience the same unexpected behaviour from the printer during mesh bed levelling:
When the mesh bed levelling is running, motor feedback is silently used as 'sensor inputs' for the mesh levelling system (Z stop reached, X Y origins reset).
In normal cases this should never happen. But when the printer is malfunctioning, due to high friction by a skewed trapezoidal nut for example, the mesh bed levelling does not detect this unexpected input and just continues resulting in a inconsistent bed level and a user that is not informed by the error.
As @CarlosGS suggested: a warning would be nice. For example that the bed levelling is triggered by the stepper motor driver sensor less probing feature, and not by the superPINDA. Or that half way during mesh bed levelling suddenly X and Y axis origins are reset.

Prusa-Support, 2022-07-20:
Sure, that's understandable. Such buggy behavior should probably be observed and studied further, and it is already being discussed in #1131.
For this reason, I assume I can close this issue.

Please feel free to participate or follow to the issue mentioned above.
I hope you don't mind me closing this issue. Please let me know if there is a need to reopened it.

Michele Moramarco
Prusa Research

closed 2022-07-20 by Prusa-Support