prusa3d/Prusa-Firmware-Buddy #1358: [BUG] Prusa Connect Local doesn't work

vasimi, 2021-04-07:
**Printer type** - MINI

**Printer firmware version**-  4.3.1

**Original or Custom firmware** Original

**Optional upgrades** - Filament Sensor

**USB drive or USB/Octoprint**
USB flash drive

**Describe the bug**
When I turn the printer on, Prusa Connect Local works just fine. However, after a while, I'm not able to load the website, it timeouts. Turning the printer off and on again helps and Prusa Connect Local works again for a while.

**How to reproduce**
Connect LAN & wait

**Expected behavior**
Prusa Connect Local works all the time

**G-code**
not related to a specific g-code, this issue occurs even if the printer is not printing

**Crash dump file**
no crash dump

**Video**
nope

reedy, 2021-04-12:
May be related to, or a dupe of #1285

vasimi, 2021-04-12:
@reedy I don't think it is, the linked issue only mentions `main.js`, I can't even load the index file (and it's not just slow, it doesn't load at all).

reedy, 2021-04-12:
Some people did report on that bug that it never loads at all on some Browsers

der1igedon, 2021-04-14:
I've got the same problem. Any ideas yet? Thanks!

menthol1979, 2021-04-17:
Same problem here. Switched to Octoprint.

jdaandersj, 2021-06-02:
I am having the same problem, seems arbitrary but I have had js, css, and index fail separately. So far have not had all load successfully. Have tried chrome + firefox, also multiple reboots of printer.

closed 2024-06-17 by github-actions[bot]