prusa3d/Prusa-Firmware-Buddy #1364: [FEATURE REQUEST] Enable EMERGENCY_PARSER

hasechris, 2021-04-11:
https://github.com/OctoPrint/OctoPrint/issues/3699

Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.
## search and found nothing ##

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MINI]
**Optional upgrades** - none

**Is your feature request related to an existing problem? Please describe.**
Printer Dialouges wont be processed on Octoprint because of not enabled EMERGENCY_PARSER Capability.

**Describe the expected functionality**
With the Capability EMERGENCY_PARSER we can send M-Codes like M876 (Dialoge Handling) to the printer while it is sending "busy:xx" Messages to the octopi.

hasechris, 2021-08-24:
sorry but Bump. This feature is really needed for good octoprint handling...

Thynix, 2022-09-17:
As of this writing, enabling `EMERGENCY_PARSER` encounters

```
#error "EMERGENCY_PARSER is not yet implemented for STM32F4/7. Disable EMERGENCY_PARSER to continue."
```

Prusa-Support, 2022-09-17:
Hello! Thank you for the report, sorry for the late response from my side. From what I could see internally, this aspect is known to us and it is currently in the works. In case of more info on this, we'll update on this thread, although I can't guarantee a possible timeframe for now. 

Alessandro Pantaleo
Prusa Research

hasechris, 2022-09-17:
> As of this writing, enabling `EMERGENCY_PARSER` encounters
> 
> ```
> #error "EMERGENCY_PARSER is not yet implemented for STM32F4/7. Disable EMERGENCY_PARSER to continue."
> ```

Hi :-)

maybe the error is only for the version in this repo? In this octoprint issue [0] someone mentions that there seems to be support for EMERGENCY_PARSER in STM32. But sadly i dont have more information regarding this so i could be wrong.

[0] https://github.com/OctoPrint/OctoPrint/issues/3699#issuecomment-729134784

hasechris, 2022-09-17:
> Hello! Thank you for the report, sorry for the late response from my side. From what I could see internally, this aspect is known to us and it is currently in the works. In case of more info on [...]

Hey Alessandro,

that sounds really cool :-) Would be so great to use my mini+ remotely from octoprint.

> sorry for the late response from my side.
happens from time to time :-) But considering the amount of issues and the response times from Prusa (really, not you. I'm not mad or anything) here on github maybe prusa could devote more workhours from you guys into the mini+ repo management? That would help immensly.

Best regards
hasechris

hasechris, 2024-05-21:
ugh - stale bot things. @Prusa-Support please remove the stale label - you said you are working in this?

btw: stale bot is really not helping. Just shoving a wall into the faces of the community with the label "issue is stale" is really bad if you are the cause for stalling.

Prusa-Support, 2024-07-13:
Thanks for commenting and keeping the issue active.

It wouldn't be possible to identify the most meaningful issues that deserve more attention without the stale-bot and we are not in the position to decide what request deserves priority - or, to better say it, hardly [as you can see by the number of issues among active or not](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues?q=).

The community contribution in carrying on community requests can be appreciated from little actions like adding a comment when the stale bot kicks in, so a simple comment would be all. Thank you.
Yet, please notice that a closing action would be easily _reversible_. We keep monitoring.

MIchele Moramarco
Prusa Research

hasechris, 2024-09-12:
ugh - stale bot things.