prusa3d/Prusa-Firmware-Buddy #1374: [ENHANCEMENT]<Better Steel Sheet Naming also showing offset>

raymate, 2021-04-22:
If possible change the HW -> Steel Sheets (labels) to something that represents the current offered sheets from Prusa so use the naming that relates to the product on the print bed.

It would also be very helpful to also append to the each sheet that is configured the name followed but the current offset value

**Example:**

Smooth 1 -0.985
Smooth 2 -1.150
Satin 1 -1.560
Satin 2
Textured 1 -1.855
Textured 2

This will help when your under settings screen you will see the current selected sheet (this caused me some confusion as I have a number of sheets and it said Custom1 and I actually could not remember which sheet this was in my collection)

Also add a label to the setting screen so it would say under the screen:

**Example**

Temperature
Sheet: Satin 1 -1.560
Move Axis
etc.

Danafra, 2021-06-03:
I second this request!

chrishilbert, 2021-06-08:
Third

Eldenroot, 2023-01-11:
Me too - displaying Z offset could help a lot .)

Prusa-Support, 2023-01-31:
Hello! Thank you for the idea, nothing I can promise for now but I brought it to devs, it's possible though not certain we can have it in the future. 

Alessandro Pantaleo
Prusa Research

CZDanol, 2024-05-16:
Closing in favor of #1318

closed 2024-05-16 by danopernis

CZDanol, 2024-05-16:
Just to be clear - the idea is that if we add support for custom sheet naming, you'll be able to add the number there yourselves.