prusa3d/Prusa-Firmware-Buddy #1377: [BUG] USBH_Thread. Precise data access error, precise

itsthejoker, 2021-04-22:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version**-  4.3.0+2756

**Original or Custom firmware** Original

**Optional upgrades** - None

**USB drive or USB/Octoprint** - USB

**Describe the bug**
When loading new filament, I removed the USB stick while waiting for the purge to finish and the printer immediately crashed. I attempted to replicate after rebooting the machine by removing and re-inserting the USB stick while the loading process was running, but was unable to trigger the crash again.

**How to reproduce**

1. Begin the filament load process with USB stick inserted
2. At the screen where it asks you if it's the correct color now, select "No" to continue extruding
3. Remove USB stick during extrusion process
4. (maybe?) Observe crash

**Expected behavior**
The printer will continue extruding without crashing.

**G-code**
N/A

**Crash dump file**
[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/6361321/dump.zip)

**Video**
I don't have a video, but I do have a picture of the exception screen.
[image]

JohnnyDeer, 2021-04-26:
Hi @itsthejoker, this bug is fixed in latest fw version 4.3.2-beta1. Should not happen anymore.

JohnnyDeer, 2021-05-13:
I´m closing this issue as solved.

closed 2021-05-13 by JohnnyDeer