prusa3d/Prusa-Firmware-Buddy #1383: [BUG]Running 2 MINI+s on LAN and they keep forgetting there hostnames.

waltercwitt, 2021-04-27:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI+

**Printer firmware version**-  4.3.1

**Original or Custom firmware** -  Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
USB flash drive

**Describe the bug**
  Both printers have hostnames set, prusamini1 and prusamini2. Sometimes after restarting they will forget there hostnames and 
  I have to reset them.

  I have also run into the following issues while troubleshooting.
 - If, during power on, there is a flash drive that has a lansettings.ini file on it plugged into the machine, It would overwrite it's 
   current settings with the ones in the file.
 - When I save the LAN settings to a flash drive, the hostname it saves is always blank, even when the hostname is set and 
   working.
 - When I load the LAN settings from a flash drive using the menu, it very rarely actually works, meaning I have to use the 
   bootup method in my first point to fix it.

**How to reproduce**
  Have 2 Mini+s connected to the same LAN network, and assign a hostname to each. after a few restarts, (longer time off seems 
  to make it more likely) one or both will forget there hostname. this is at work so I cannot make changes to the network 
  configuration.

**Expected behavior**
Covered in more detail on my forum post: https://forum.prusaprinters.org/forum/hardware-firmware-and-software-help/problems-with-hostnames-in-firmware-4-3-1-using-more-than-one-prusa-mini/

**G-code**
  None

**Crash dump file**
  None

**Video**
  None

JohnnyDeer, 2021-05-06:
Hi @waltercwitt, thank you for reporting. We are aware of this issue, unfortunately have low priority at this moment. Thank you for understanding and patience.

kenzon, 2023-11-29:
We have same kind of problems with this device:

Printer type - MINI
Printer firmware version - 5.1.0
Bootloader version - 2.3.0
Buddy Board - 105

The problem is that the menu on the Prusa MINI does not have an entry where we can change the hostname of the printer.
The hostname being the name there is seen on the network and being the first part of the FQDN for the printer.

Thus adding a new printer on the network just gives more problems with yet another machine there calls it self for “prusamini”, the more we add the more havoc we have in our network.

Why is this still not possible, it is a quite normal setup feature for almost all IoT devices ?

Please regard this a feature request.

kenzon, 2024-06-17:
Does this mean that you are not willing to make the improvement?

danopernis, 2024-06-17:
It means it is not a priority, because nobody cares enough to even comment on it for 7 days to prevent automatic closing :shrug:

sl1pkn07, 2024-06-22:
this should remain open and fixed with add a option in the ini file

murk-sy, 2024-06-23:
I have 4 minis on the network and the router is reporting different hostnames for each, as I've set them, currently on 6.0.2.

Have you tried reproducing the issue actually?

Here's example of a Prusa Connect generated ini:
```ini
[printer]
type = MINI
name = Mini 01
location = x

[network]
hostname = mini-01

[service::local]
enable = 1
username = 
password = 
api_key =
```

sl1pkn07, 2024-06-23:
ok, thats is i looking for

thkanks

Prusa-Support, 2024-09-14:
> The problem is that the menu on the Prusa MINI does not have an entry where we can change the hostname of the printer. The hostname being the name there is seen on the network and being the first part of the FQDN for the printer.
> 
> Thus adding a new printer on the network just gives more problems with yet another machine that calls it self for “prusamini”, the more we add the more havoc we have in our network.

Starting from [FW 6.1.2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.1.2), the user can customize the hostname directly in the printer's Network (/Wi-Fi or /Ethernet) menu thanks to the new alphanumeric keyboard on the screen.

Despite being old and forgotten (only apparently), I wouldn't mind reopening the issue since it may still seem to be relevant for some of you, however, _I believe we may consider the issue solved by customizing the hostname on the screen_.

This issue will be opened for now and I'll wait for your feedback before potentially closing again as officially solved.

Michele Moramarco
Prusa Research

kenzon, 2024-09-17:
Thank you Prusa-Support,
we have just upgraded our mini's to Firmware 6.1.2

Now we finally have the very nice feature where we can customize the hostname directly in the printer.
This will eventually solve our problems we have in our network. 👍

closed 2024-09-17 by CZDanol