prusa3d/Prusa-Firmware-Buddy #1385: [BUG] Connecting Octoprint in "print finished" window doesn't block printing from USB drive

DangeMask, 2021-04-28:
**Printer firmware version**
4.3.1 Original

**USB drive or USB/Octoprint**
  USB drive + Octoprint

**Describe the bug**
  Usually when I connect Octoprint to my prusa MINI, the display shows Octoprint logo and menu is reduced to tuning only.
Today I printed one print from USB drive (needed a stop in print) and then without leaving "print finished" window, I connected Octoprint and started another print. Later I discovered, that the display on MINI still shows "printing" dialog, allowing me to go to "Print" menu and select another gcode...
I have waited for the print to end and selected  printing from USB drive, the printer went on, but Cctoprint still remained connected. I was even able to start another print from Octoprint with no error.

The result of two consecutive prints running was this:
- printer heated up to 170°C, did mesh bed leveling
- waited for nozzle temperature 220°C
- printed purge line and the skirt for "USB drive" model
- stopped moving and waited for temperature to drop to 170°C again. (I assume for mesh bed leveling of "Octoprint" model
At that point, I stopped it to not do any harm.

**How to reproduce**
  Finish print from USB drive, DO NOT CLICK ANYTHING ON MINI, connect Octoprint.

DangeMask, 2021-05-07:
UPDATE: Whem I'm in "tuning" menu (maybe any other than home window) and try to connect Octoprint, I get the same intermediate state, when the printer allows me the whole menu and Octoprint is connected, allowing me to print.

JohnnyDeer, 2021-05-12:
Hi @DangeMask, thank you for reproting.

DangeMask, 2021-05-21:
UPDATE: the same goes to disconnection. 
Reproduce steps:
1) Connect Octoprint
2) Go to "Tune" mneu in MINI
3) Disconnect Octoprint via web interface
4) Leave TUne menu in MINI
The screen still shows "SERIAL PRINTING" and to access all features, you have to select "Disconnect" in MINI too.

closed 2024-06-17 by github-actions[bot]