prusa3d/Prusa-Firmware-Buddy #1387: [ENHANCEMENT] Add confirm menu for Change Filament option while printing

Combinacijus, 2021-04-29:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MINI
**Optional upgrades** - Filament Runout Sensor

**Is your suggested improvement related to an existing problem? Please describe.**
It's not the first time when I play with `Live Z Adjust` or `Speed` options while printing and I accidentally press `Change Filament` which unloads filament immediately with no way to stop it

**Describe the expected functionality**
When `Change Filament` is selected open confirmation menu before unloading filament
or 
Add abort option while Z axis is moving up and still haven't started to unload filament

tverett, 2021-06-03:
Agreed this would be a big improvement. Also while you’re enhancing this menu item add a sheet profile change request/prompt too so it doesn’t get forgotten.

imnotme, 2021-07-01:
Please, so much frustration can be avoided with this option, I even think less printer will be harmed if this get implemented.

jwidess, 2024-03-28:
I've done this countless times, really hope Prusa adds this simple confirmation option someday. 😄

jwidess, 2024-04-10:
The [Snake Mini](https://github.com/Snake-Edition/P32-FW) Firmware now has a version of thus functionality for anyone interested. It was just added in the [Snake MINI FW 5.1.2.2](https://github.com/Snake-Edition/P32-FW/releases#:~:text=14-,Change%20filament%3A%20Change%20of%20filament%20in%20Tune%20menu%20is%20moved%20to%20submenu%20to%20avoid%20unwanted%20interruption.,-18%20Speed%20up) release today.

closed 2024-07-04 by github-actions[bot]