prusa3d/Prusa-Firmware-Buddy #1393: Extruder first layer mess

PowerChaos, 2021-05-01:
Prusa Mini 4.3.2-Beta and 4.3.1 Stable

i was printing stuff for some time now , and suddenly the first layer of the mini started to mess up
it extrude more then 500% layer height with a extreme slow speed ( normal speed of 30mm/s , it extrude at a speed of 5 mm/s ) 

if i enable Relative Value, then it is fine
if i enable Absolute value the mess begins

my guess, the mini does not work that nice with Absolute value's and need relative value's to work

same problem with prusa slicer
test files are done with SimplyFy 3D and prusa slicer 2.3.1 ( bondtech profile , based on orginal profile -> nothing changed exept bondtec e steps ) 

[image]
[image]

[hartje.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/6409658/hartje.zip)

Zip file is the gcode , the Absolute does provide that thick layer , the relative verison does work nicely

Greetings from PowerChaos

JohnnyDeer, 2021-05-06:
Hi @PowerChaos, this issue is caused by Gcode, specifically by combination of M82 and M83 instructions as you can see on screenshot. 

[image]

After changing second instruction to M82 Gcode work as expected. 
I´m closing this issue as invalid because is not related to FW.

closed 2021-05-06 by JohnnyDeer

PowerChaos, 2021-05-06:
Thank you

So in this case it is related to the slicer ( simplify 3d and prusaslicer ) and the single button that says “relative” ( disable for absolute mode ) that cause the gcode to mess up after using the same settings for about 2 months ?

for testing purpose some one else sliced a benchy for me in absolute mode with prusa slicer
Result was that the skirt was fine but the benchy got under extruded a lot ( somehow )  switched to relative mode and it was fine 

as Sidenote. The first layer test ( Build in Z pattern ) is perfect. So first layer is set correctly.

I never manual edit the gcode and only used the slicer 
Only on the mini I added 2 values as startup for the bondtec. Everything else is default

to rule the problem out. Can you slice me 2 files with default settings for the prusa mini?
One in relative mode and other in absolute mode. Then I going post here the 2 screenshots of the first layer so that slicer problem can be ruled out ( as it worked fine with same settings before )

it would be strange that 2 differend slicers provide exact same result or same “broken” gcode on default profiles 

thank you
Greetings from PowerChaos