prusa3d/Prusa-Firmware-Buddy #1395: [BUG] BSOD after upgrading to 4.3.2-beta 1

cmumford, 2021-05-03:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version**-  4.3.2-beta 1

**Original or Custom firmware** Original

**Optional upgrades** - Filament sensor

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint

**Describe the bug**
  After upgrade my MINI goes into a boot reset loop where it emits a loud beep, a BSOD before resetting. Booting w/o a flash disk in the USB port still results in an error. Also after a complete powerdown.

**How to reproduce**
  Started with 4.3.0
  1. start MINI.
  2. Insert USB drive containing Buddy_4.3.2-beta+2736.bbf.  (SHA256 hash: AD06E1FD4CB6ECE044D6203BCE2B0691674544209F5DC2F7E24E1FB4E48679FD)
  3. Hit reset button on MINI.
  4. MINI upgrades the firmware.
  5. Mini displays BSOD after upgrade.
  6. Mini resets.
  7. Back to step 5.

**Expected behavior**
  Mini will start as normal.

**G-code**
  N/A failure on boot.

**Crash dump file**
  Not sure how to get this, but will attach BSOD photo.

**Video**
  LMK if photo is insufficient and video is desired.

[image]

cmumford, 2021-05-03:
I just realized that I had the mini-USB port connected to a computer running Octoprint. If I unplug the cable then the MINI boots just fine, but **fails with the cable plugged in**.

JohnnyDeer, 2021-05-05:
Hi @cmumford, this issue is not related to upgrading printer. It´s caused by μ USB connected to computer. We are aware of this bug and will fix it in future.

JohnnyDeer, 2021-05-25:
New FW version with fix of this bug is out. @cmumford please try it out ad let know if this issue can be closed :)
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.2-RC1

cmumford, 2021-05-25:
Just installed v4.3.2-RC1 and it flashed and booted just fine - no errors. I also connected via OctoPrint, and it works as well. I'm doing a quick print now via OctoPrint. It's not done, but so far all is ok. 👍

closed 2021-05-25 by JohnnyDeer