prusa3d/Prusa-Firmware-Buddy #1400: [BUG] If filament runs out during print, new filament defaults to PLA

kosme, 2021-05-06:
**Printer type** - [MINI]

**Printer firmware version**-  [4.3.1]

**Original or Custom firmware** [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint**
  USB flash drive

**Describe the bug**
  If a piece is being printed and the filament runs out, the print will stop, the printer will ask for a new filament to be loaded but will default it to PLA. The print continues with the correct settings for temperature, but the screen shows that the loaded filament is PLA even if it is not.

**How to reproduce**
  1. Load a short piece of a filament of a material different than PLA and start a print. 
  2. Let the print stop because the filament finished. 
  3. Follow the screen instructions and load a new filament of any kind.
  4. The screen will now display that the new filament is PLA.

**Expected behavior**
  When loading a new filament during an interrupted/paused print, the machine should ask the material of the new filament.

JohnnyDeer, 2021-05-19:
Hi @kosme, tested on 4.3.1 2759 with PETG PVB HIPS and not reproduced with and without factory reset after FW upgrade. Could you please try it again with fresh FW update and factory reset?

kosme, 2021-05-20:
@JohnnyDeer Will give it a try

kosme, 2021-05-29:
@JohnnyDeer I tried again and the bug persists. I include a link to a video that shows it [here](https://youtu.be/8zG28wsg25Q)

JohnnyDeer, 2021-06-02:
@kosme Thank you for video, I will investigate it more.

JohnnyDeer, 2021-06-03:
I did reproduce it, happens after booting printer with already loaded non PLA filament.

murk-sy, 2021-06-15:
Any ETA on the fix? First layer calibration is kind of crippled with this and requires a manual preheat + filament change before you can use it.

reedy, 2021-06-20:
Worth mentioning it's the case on 4.3.2 aswell

mrstoned-gh, 2022-04-04:
Can also mention bug is still in 4.3.4 with a filament color-change using prusaslicer.

To reproduce;
Load any other then PLA.
Slice a model in PS with a  colorchange.
After colorchange, display now shows PLA.

Prusa-Support, 2022-06-24:
Thank you all for your feedback.

This is a known bug that has not yet been fixed, so this is a work in progress.

Thanks for your patience and cooperation.

Michele Moramarco
Prusa Research

reedy, 2022-08-07:
[image]

So is this actually just a display issue? It says it's pla, but it's not. Or is it suddenly treating it like it's PLA?

murk-sy, 2022-08-08:
I have had a case on 4.4.0-BETA2 where PETG ran out and new one was loaded, and on next print the material warning popped up. Printer still says PLA loaded.
It is possible this is becaue PETG has been loaded since before firmware update, but will keep an eye out.

reedy, 2022-08-08:
[image]

Seems it doesn't always happen? Or only happens under certain circumstances?

Printer was on PLA from the last print (but petg loaded) at start up. Changed to petg with change filament, did a print with two filament changes (both petg) and it's still showing petg...

Prusa-Support, 2022-08-13:
The condition for the bug to occur should be: 
- the filament runs out during the print;
- a color change is prompted via gocde (set in advance in Prusa Slicer).

I'm unable to tell if there are circumstances under which the bug is not persistent, but I believe it doesn't occur if you change the filament at any point during the print, via 'tune' menu.

Unfortunately, the bug is not fixed yet.

The bug won't cause issues or hazardous situations: tthe print will continue as per the gcode direction, without temperature modifications.
As a direct consequence of the bug, the printer will prompt an incorrect "wrong filament" warning on the next print - unless the user changes the filament via 'change filament' option, or simply prints PLA (because the printer expects gcodes sliced for PLA as a consequence of the bug).

Michele Moramarco
Prusa Research

murk-sy, 2022-08-15:
Another (biggest in my opinion) consequence of the bug is that if you wish to change filament afterwards, you need to manually preheat the hotend to the appropriate temperature, otherwise it tries to change it at PLA temperatures. A recipe for grinding filament if you're not careful.

murk-sy, 2022-08-19:
Something I just thought of: For me, filament runout generally happens on overnight prints when printers are unattended.

It is possible this is related to the heating timeout that occurs 30 minutes after filament is removed and printer is idle - but can't say for certain just yet.

reedy, 2022-08-19:
I've had it when I've changed the filament within a couple of mins of it needing changing

Prusa-Support, 2022-08-30:
The bug occurs if the filament runs out or a color change is prompted via G-code. Some hassles during the filament change are sadly the consequences. 
So far, it doesn't seem to occur in other circumstances but we have enough clues for our developers to look into this.

Michele Moramarco
Prusa Research

Prusa-Support, 2022-10-15:
This issue is fixed with the new release candidate firmware [4.4.0-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-RC1).
>Incorrect filament displayed after a filament runout. If the user loaded a PETG filament, restarted the printer and started a print during which a filament runout occurred, the printer then shows PLA as the loaded filament. This is now fixed.

This is a release candidate, which means safe enough very close to a stable release so I'll close this issue now.
Feel free to help us test and please let us know if you think this issue should be reopened for some reason.

Michele Moramarco
Prusa Research

closed 2022-10-15 by Prusa-Support