prusa3d/Prusa-Firmware-Buddy #1429: Please th show the sheet profile be shown in the calibration and print screens

PaulDC, 2021-05-19:
I think the different sheet profiles is an excellent feature. However I've often forgotten to change the sheet profile when changing to a different sheet texture. I've usually realised this just after the print has started and then had to abandon the print to change the sheet profile.

Please would be it be possible to show the sheet profile selected  in the Calibration screen and the Live print screen ? I would then  realise before the print has started that I've not changed to the correct sheet profile.

Many thanks,
Paul C.

JohnnyDeer, 2021-05-20:
Hi @PaulDC thank you for suggestion.

chillygithub, 2021-05-23:
Can I suggest adding it to the "Change Filament" menu. As this is when you typically also need to change the sheet.

tverett, 2021-06-03:
Agree to both requests mentioned here. Showing on main screen and grouping with change filament.

darraghbr, 2022-06-11:
Another way to do this might be to interrogate the user every time a print is started which sheet is installed. This might be especially helpful when multiple people are using one printer. Clearly this would be something that the user could enable and disable as some would find it excessive.

Prusa-Support, 2022-06-21:
Hello! Thank you for your ideas and sorry for the late response from my side. All noted, still under consideration from our side - sadly, no guarantees yet that anything would be implemented, but we'll consider it. 

Alessandro Pantaleo
Prusa Research

closed 2024-06-17 by github-actions[bot]