prusa3d/Prusa-Firmware-Buddy #1437: Prusa Mini (4.3.1) reports Probing error whe enter

saysphilippe, 2021-05-24:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version**-  [e.g. 4.0.5, ...]

**Original or Custom firmware** [Original/Custom]

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint

**Describe the bug**
  A clear and concise description of what the bug is.

**How to reproduce**
  Please describe steps to reproduce the behavior.

**Expected behavior**
  A clear and concise description of what you expected to happen.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

JohnnyDeer, 2021-05-25:
Hi @saysphilippe thank you for creating issue. I wanted to ask you upload video which you already did. From video is not obvious same behavior as on MK3S right before M112 but i think that its caused by same thing, broken bed thermistor cable. Here is MK3S issue related to this one https://github.com/prusa3d/Prusa-Firmware/issues/3083. Also please contact our Tech Support for help with link to this issue (chat in right down corner https://shop.prusa3d.com/en/).

saysphilippe, 2021-05-27:
Both Prusa and Octoprint reports Probing error, but I guess I will have to order a new bed thermistor cable, then.

saysphilippe, 2021-05-28:
Question: Why would a bad bed thermistor cable only react when probing, and never when printing?

JohnnyDeer, 2021-06-02:
@saysphilippe Because often cable need to be in specific angle to cause disconnect, even just for half of second. In this case end row on left side which normally is not area where you print often, isnt? May be this can also be caused by broken cable in PINDA, please discuss it with Tech Support. But definitely this is not caused by FW bug.
You don't need to order it, our Tech Support will help you :)

Prusa-Support, 2022-05-20:
Hello @saysphilippe! Sorry for intervening on this so late. Was this issue resolved in the end through our Technical Support? Let me know, in case it's still not solved we can review it in more detail with me via email (info@prusa3d.com mentioning me "attn: Alessandro). 

Alessandro Pantaleo
Prusa Research

bfeist, 2022-10-30:
I replaced the thermistor for my heatbed and the problem persists. M112 at the end of every print.

Prusa-Support, 2022-10-31:
Hello! Thank you very much for the update, good to hear it is solved, sorry for this taking the unnecessary replacement and a long time. Good to have this feedback, we'll keep it in mind if we come across similar cases. 

For the future, in case of something that can be tested-troubleshot, I'd always recommend contacting us, technical support, directly: info@prusa3d.com, even me directly at that email address, mentioning me in the subject. We are trying to catch up with the threads on this GitHub, but it's meant more for development aspects.  

Alessandro Pantaleo
Prusa Research

closed 2024-06-17 by github-actions[bot]