prusa3d/Prusa-Firmware-Buddy #1440: [QUESTION]Where is the source code for the actual bootloader shipped with the Prusa Mini+?

matthewlloyd, 2021-05-26:
The Prusa Mini+ Buddy board ships with a bootloader pre-flashed at the factory.

When compiling the firmware source code from this repository, the bootloader is downloaded in binary form by `utils/bootstrap.py` as part of the build process, from this URL:

https://prusa-buddy-firmware-dependencies.s3.eu-central-1.amazonaws.com/bootloader-mini-1.0.0.zip'

This ZIP file contains a 128KB binary file named `bootloader.bin`:

```
Archive:  bootloader-mini-1.0.0.zip
  Length      Date    Time    Name
---------  ---------- -----   ----
   131069  2019-11-08 07:09   bootloader.bin
      212  2019-11-08 07:09   __MACOSX/._bootloader.bin
---------                     -------
   131281                     2 files
```

The source code which generates this binary file, which contains executable ARM code, does not seem to be available anywhere in Prusa's GitHub repositories. This bootloader is distributed with the Prusa Mini+, in the flash memory on the ARM chip on the Buddy board, alongside and packaged together with the rest of the firmware, which of course is based on Marlin.

The entire project is a covered work under the terms of the GPL v3 license, as it is based on Marlin which is also GPL v3. I believe that Prusa is required to supply the complete source code for the entire firmware, including the bootloader, because it is packaged together in the distribution medium which Prusa itself distributes, to form a larger program. Therefore, in order to comply with Marlin's GPL, I believe Prusa is required to supply the source code for the bootloader.

I hereby request Prusa make available the source code for `bootloader.bin`.

JustAnother1, 2021-06-04:
It is sad that prusa does not comment on this :-(

They might think that as long as they do not link the bootloader against the firmware they don't need to provide the source. One could argue that the bootloader and the firmware are separate executables that are side by side in the chip and therefore different software. If that holds then they would not be required to publish the source. But I'm not a lawyer.

But they also release the 3d files of the plastic parts and so much other stuff that they would not need to release. So why not the bootloader? Maybe they have a good reason?

If you do `$ strings bootloader.bin` you get all the text contained in the binary file.
[strings.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/6599865/strings.txt)

So the Bootloader can do an update from USB. And there is this string 

> Developer firmware

Maybe they don't want everybody to know how to create a Developer firmware ?

Next we can do this 
`$ arm-none-eabi-objdump -D -bbinary -marm bootloader.bin -Mforce-thumb >  output.s` 
that gives us a disassembly.
[output.s.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/6599897/output.s.zip)

Looking through that could take some time. A help could be Radare2 like this 
`$ r2 -a arm -b 16 -m 0x08000000 -w bootloader.bin ` 
To understand how that works take a look here: https://radare.gitbooks.io/radare2book/content/first_steps/intro.html

So now who will be first to understand how the Developer firmware feature works?

matthewlloyd, 2021-06-05:
> They might think that as long as they do not link the bootloader against the firmware they don't need to provide the source.

GPLv3 is quite clear on this topic:

> 
> 5. c) You must license the entire work, as a whole, under this License to anyone who comes into possession of a copy. This License will therefore apply ... to the whole of the work, and all its parts, regardless of how they are packaged.
> 
> A compilation of a covered work with other separate and independent works, which are not by their nature extensions of the covered work, and which are not combined with it such as to form a larger program, in or on a volume of a storage or distribution medium, is called an “aggregate” if the compilation and its resulting copyright are not used to limit the access or legal rights of the compilation's users beyond what the individual works permit. Inclusion of a covered work in an aggregate does not cause this License to apply to the other parts of the aggregate.

The first paragraph clearly applies. Prusa might argue that the combination of the bootloader and the main program is just an aggregate, but the license clearly states that it is not an aggregate if they are combined to form a larger program in or on a storage or distribution medium - and clearly the flash memory of the STM32F407 is a distribution medium. Furthermore, Marlin includes bootloaders, so the Prusa bootloader is "by its nature an extension of the covered work".

> Developer firmware

Good find. Some disassembly reveals the "Developer firmware" message is activated if the 64-byte signature (normally produced from Prusa's private key, and verified using the public key in the bootloader) is filled with 0xFF. The signature is flashed at the start of the firmware image, at memory offsets 0x8000000 - 0x8000040. This is easy to arrange for with a small modification to the build scripts. I'm not sure what the practical implication of this is. I already broke the appendix so I can't determine whether it allows the normal intact-appendix signature check to be bypassed or not, but it's a possibility. It may also bypass the SHA hash check.

[image]

Practically speaking, I'm currently trying to get Klipper running, and it would be a much easier task if I could see how the bootloader works. Another reason I could use the source is that the bootloader that shipped on my Mini+ was v1.1.0, but the binary bootloader downloaded by the build script is an old version v1.0.0; I've already accidentally downgraded and don't have a copy of the v1.1.0 bootloader binary to replace it, let alone the source.

I hope they take care of this matter quickly, otherwise it may be necessary to escalate to their legal team, or the great man himself.

JustAnother1, 2021-06-05:
I used a original BBF file and changed the first 64 Bytes to 0xff. I could not flash that image. I did not see the Developer Firmware message.

I have bootloader 1.1.0 and did not break the appendix.

Could you share the BBF file that got you to "Developer firmware" ?

Regarding Klipper: The bootloader will most probably only shift the start address of the firmware and the address of the Vector table. You can probably get both from the Lionker file used to compile the Prusa Firmware. With those changes it should be possible to put the Klippy Firmware into a BBF file and have it flashed and run as Develeoper firmwware.

But the you will need to repeat that with every Klipper update that chenges the firmware,...

Or as you have already broken the apendix, jsut use the SWD interface, get rid of the Bootloader and flash the Klipper firmware using SWD.

You might want to sdisassemble the display unit and remove the foil from the screen, that will also remove that green thing in the bottom left corner.

matthewlloyd, 2021-06-05:
> I could not flash that image.

That confirms breaking the appendix (or lifting BOOT0 etc.) is still necessary to open things up, even when uploading "Developer firmware".

> Could you share the BBF file that got you to "Developer firmware" ?

Don't have it handy but it was identical to what you created by patching the first 64 bytes.

DFU via USB seems more reliable than the stock bootloader (slow and occasionally ignored BBF updates on the USB drive). I added a cable to extend the 2-pin header outside the box, I just add the jumper and reset, then use dfu-util. SWD debugging is working too using OpenOCD.

I have Klipper running - bootloader offset was the issue. Foil on the screen is to protect it from the appendix-shaped PCB projectile flying around the room ;)

JustAnother1, 2021-06-05:
> > I could not flash that image.
> 
> That confirms breaking the appendix (or lifting BOOT0 etc.) is still necessary to open things up, even when uploading "Developer firmware".

Not so fast. I used a different Bootloader than what you tested with. So it might be a new feature in the 1.1.0 version. They might have changed the Developer Firmware signature to something else or removed the feature completely. 

You can test if the "appendix" is the issue by connection BOOT0 and SWDIO to GND. That is all the appendix does.

Now I also want to see the Source code of the Bootloader,....

JustAnother1, 2021-06-07:
@ @JohnnyDeer What is the Prusa position regarding GPL and the Bootloader?
 
> GPLv3 is quite clear on this topic:
> 
> > 1. c) You must license the entire work, as a whole, under this License to anyone who comes into possession of a copy. This License will therefore apply ... to the whole of the work, and all its parts, regardless of how they are packaged.
> > 
> > A compilation of a covered work with other separate and independent works, which are not by their nature extensions of the covered work, and which are not combined with it such as to form a larger program, in or on a volume of a storage or distribution medium, is called an “aggregate” if the compilation and its resulting copyright are not used to limit the access or legal rights of the compilation's users beyond what the individual works permit. Inclusion of a covered work in an aggregate does not cause this License to apply to the other parts of the aggregate.
> 
> The first paragraph clearly applies. Prusa might argue that the combination of the bootloader and the main program is just an aggregate, but the license clearly states that it is not an aggregate if they are combined to form a larger program in or on a storage or distribution medium - and clearly the flash memory of the STM32F407 is a distribution medium. Furthermore, Marlin includes bootloaders, so the Prusa bootloader is "by its nature an extension of the covered work".

Can you point us to the Bootloader repository, or does Prusa have a different view on the license terms?

dragomirecky, 2021-06-15:
Hey @matthewlloyd and @JustAnother1,

Thank you for your patience regarding this issue. To answer the primary question here: the bootloader is a separate project from the Buddy firmware. The Buddy firmware is in no way dependant on it and is fully functional without it. The bootloader isn’t part of the Buddy firmware, and we don’t distribute it as part of the BBF files on our website (as mentioned in the first comment). Therefore, we believe that seen from the point of the GPLv3 license, the firmware and the bootloader form an aggregate, and we are not obliged to release the bootloader’s source code. In the same way, the license does not require us to open source the source code of the stock bootloader pre-flashed by STMicroelectronics.

But hey, that being said, we are planning to open-source the bootloader :slightly_smiling_face:. Even though it is not a current priority, we are taking steps towards making it eventually happen.

<br>

> So now who will be first to understand how the Developer firmware feature works?

Don’t want to spoil the fun, but I have to say that there is no hidden “Developer Firmware” feature. The “Developer firmware” string you found in the disassembly appears when the firmware isn’t signed. And to flash an unsigned firmware, you have to break the appendix first. So again, no hidden feature here – even all the printers at our offices have broken appendixes.

<br>

> but the binary bootloader downloaded by the build script is an old version v1.0.0; I’ve already accidentally downgraded and don’t have a copy of the v1.1.0 bootloader binary to replace it

Thanks for pointing this out! I just made a note to update the bootloader there.

JustAnother1, 2021-06-15:
> Thank you for your patience regarding this issue. To answer the primary question here: the bootloader is a separate project from the Buddy firmware. The Buddy firmware is in no way dependant on it [...]

One might argue that the Prusa Bootloader and the Prusa Firmware share the same medium(Flash) but the ST Bootloader is in ROM, but I'm not a lawyer.

 
> But hey, that being said, we are planning to open-source the bootloader slightly_smiling_face. Even though it is not a current priority, we are taking steps towards making it eventually happen.

It would be great if you could post the link to that (once it happened) here, so that others with the same question can find the link.

> > but the binary bootloader downloaded by the build script is an old version v1.0.0; I’ve already accidentally downgraded and don’t have a copy of the v1.1.0 bootloader binary to replace it
> 
> Thanks for pointing this out! I just made a note to update the bootloader there.

A link to the binary would also be appreciated (for the same reasons).

Thank you for looking into this. Like I said before Prusa is opening up so much,also opening up the mechanical (printed) parts of the printer, so not opening up the Bootloader sounded wrong. I'm happy that this will be solved. Keep up the great work!

dragomirecky, 2021-06-21:
> A link to the binary would also be appreciated (for the same reasons).

Here it is :) https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/1482

matthewlloyd, 2021-06-21:
Many thanks to Prusa for replying to this thread.

> we don’t distribute it as part of the BBF files on our website

I stand corrected, and agree that the bootloader is not distributed in the BBF downloaded from Prusa's website. It is, however, distributed on the board's flash. It is also clearly intended to be linked with it as part of the process in `utils/build.py`.

> The Buddy firmware is in no way dependant on it and is fully functional without it... the firmware and the bootloader form an aggregate.

This is not entirely accurate. The bootloader and Buddy firmware communicate bidirectionally via shared memory. Using shared memory to communicate is generally considered dynamic linking, which does make programs a combined work.

We can only see one side of the communication since the bootloader is closed source. At the very least we can see that the firmware communicates update mode flags (set via the GUI menu) to the bootloader, and the bootloader communicates the status of the appendix to the firmware. Perhaps there is more. Here are some code excerpts:

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/3b406f5cf0ba7f90471842434cbd66aed7e8434f/src/common/sys.cpp#L16

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/src/common/shared_config.h
```
typedef struct {
    uint8_t fw_update_flag;
    uint8_t model_specific_flags; // ~ "reserved1" originally
    ...
} data_exchange_t;
```

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/3b406f5cf0ba7f90471842434cbd66aed7e8434f/src/STM32F407VG_FLASH_boot.ld#L111
```
 /* to exchnage data between bootloader and application */
 /* this section is filled with zeros on each application start */
  .boot_fw_data_exchange 0x20000000 :
  {
  	_sboot_fw_data_exchange = .;  /* global symbol at start */
    KEEP (*(.boot_fw_data_exchange))
    . = ALIGN(4); /* allignment is important as 4 bytes are written to zero at once in startup code*/
    _eboot_fw_data_exchange = .;  /* global symbol at the end */
  } >RAM AT> FLASH /* AT> FLASH added to avoid the binary size increase*/
```

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/3b406f5cf0ba7f90471842434cbd66aed7e8434f/src/common/support_utils.cpp#L173
```
bool appendix_exist() {
    const version_t *bootloader = (const version_t *)BOOTLOADER_VERSION_ADDRESS;

    if (bootloader->major >= 1 && bootloader->minor >= 1) {
        return !(ram_data_exchange.model_specific_flags & APPENDIX_FLAG_MASK);
    } else {
    ...
```

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/3b406f5cf0ba7f90471842434cbd66aed7e8434f/src/common/sys.cpp#L262
```
void sys_fw_update_on_restart_enable(void) {
    ram_data_exchange.fw_update_flag = FW_UPDATE_ENABLE;
}
```

> But hey, that being said, we are planning to open-source the bootloader 🙂. Even though it is not a current priority, we are taking steps towards making it eventually happen.

This is very good to hear. I do hope given the above, that Prusa honors the open source community and respects the terms of Marlin's GPLv3 by releasing the source code for the dynamically linked bootloader as soon as is reasonably possible.

matthewlloyd, 2021-06-21:
> Here it is :) #1482

Fantastic - thank you so much for uploading the latest version. We are all very excited about the upcoming release of the i3 MK4 and XL.

[image]

miko007, 2021-06-25:
@matthewlloyd what you describe only applies, if the dynamically linked software is under gpl license. then you are correct, the linking software has also to comply with the gpl.

the other way around makes no sense whatsoever, because it would leave people unable to dynamically link non gpl libraries in their gpl projects.

i can not force, say microsoft, to open source their proprietary library by dynamically linking it in my gpl project, you see my point, right?

matthewlloyd, 2021-06-25:
> i can not force, say microsoft, to open source their proprietary library by dynamically linking it in my gpl project, you see my point, right?

The GPL specifically carves out an exception for "System Libraries", which applies for example to Microsoft's proprietary libraries, but does not apply here because a bootloader with bidirectional communication via shared memory does not meet the definition of a System Library. You can read the GPL FAQ here for more information: https://www.gnu.org/licenses/gpl-faq.html#SystemLibraryException.

The incorrect assertions in the rest of your comment are addressed in the GPL FAQ here, and in other places: https://www.gnu.org/licenses/gpl-faq.html#MoneyGuzzlerInc. In short, the GPL does not allow GPL-covered software to be linked (dynamically or otherwise) with closed source libraries, except for the System LIbrary exception as described above, and for aggregates, which this is not (see https://www.gnu.org/licenses/gpl-faq.html#MereAggregation).

miko007, 2021-07-19:
@matthewlloyd that of course was an exaggerated example.

you can not force me to put my MIT licensed js library under GPL by using it in your GPL project, either.

matthewlloyd, 2021-07-20:
> you can not force me to put my MIT licensed js library under GPL by using it in your GPL project, either.

Your statement is irrelevant because there is no MIT licensed library involved here. I have already addressed your original incorrect assertions in my previous response. My original arguments still stand. If you wish to improve your understanding of the GPLv3, I recommend reading the GPL FAQ and the text of the license itself.

miko007, 2021-07-21:
then this discussion does not lead anywhere. my original arguments also still stand and i do not need to improve my understanding of the gpl.

JustAnother1, 2021-07-21:
@miko007  The issue here is that the "dynamic linking" if we can call it that goes both ways.

1.) The Bootloader boots the Firmware. (That is the Linking that would **not** cause an issue if the Bootloader is non GPL but the Firmware is)

2.) The Firmware accesses / talks to / uses the Bootloader. After booting the Firmware the Bootloader is not just gone. As pointed out before there is interaction between Firmware an Bootloader. Therefore we have some linkage from the Firmware to the Bootloader.

Then again you are right the discussion does not lead anywhere as @dragomirecky already promised to open source the Bootloader. So if they have to do it or they do it on their own is a moot point.

matthewlloyd, 2021-07-21:
All they have stated is that "we are planning to open-source the bootloader 🙂. Even though it is not a current priority, we are taking steps towards making it eventually happen."

While I applaud Prusa's agreement to come into compliance, their statement is not sufficient under GPLv3. Since the bootloader is dynamically linked, they are _legally obliged_ to provide the source code, and furthermore, it needs to be provided _in a reasonable period of time_. It has already been _two months_ since I opened this request. "Eventually" is not good enough. It is not a question of priorities - it is a legal requirement.

When I have made similar requests to other companies, they have been handled within _days_. For example, I asked BigTreeTech to provide the source code to their modified version of RRF for their Octopus board, they did so in _two days_ (https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/issues/40). When I asked Fokoos to provide the source code to their modified version of Marlin for their Odin F5 printer, they supplied it within a _few hours_.

JustAnother1, 2021-07-21:
@dragomirecky Could you provide reasons for the delay in open sourcing the Bootloader? Keeping the  "i3 MK4" and "XL" a secret couldn't be it any more. Do you have swear words in the code that need to be removed before going public?

So it basically is just a click on "make it public" ? Or what needs to be done ? Or are you waiting on approval from @josefprusa ?

bschollnick, 2021-07-21:
> When I have made similar requests to other companies, they have been handled within days. For example, I asked BigTreeTech to provide the source code to their modified version of RRF for their Octopus board, they did so in two days (bigtreetech/BIGTREETECH-OCTOPUS-V1.0#40 <https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/issues/40>). When I asked Fokoos to provide the source code to their modified version of Marlin for their Odin F5 printer, they supplied it within a few hours.

By no means am I even related to Prusa, but…

> While I applaud Prusa's agreement to come into compliance, their statement is not sufficient under GPLv3. Since the bootloader is dynamically linked, they are legally obliged to provide the source [...]
Define reasonable period of time.

And then consider that there maybe proprietary comments, code that relies on proprietary functions or modules, and generally cleaning the code up to meet their standards.

Prusa has stated that they are working on this, so at this time, they have stated that they intend to make it available.  They haven’t given a time line, so ranting isn’t going to make it happen faster.

If you want to see it faster, then give them constructive reasons that it is worth putting more man power on that project.  

Why is it advantageous to Prusa to redirect manpower from Prusa Slicer, or other software projects to open sourcing the boot loader?
For all we know they may only have a single FTE or a partial FTE on this project.  That’s more than enough to show GPL compliance..

GPL compliance does not mean complying with your timeline, just that it will be available and that they are working on it.

	- Benjamin

matthewlloyd, 2021-07-21:
> Define reasonable period of time. ...  GPL compliance does not mean complying with your timeline, just that it will be available and that they are working on it.

I have not defined a timeline. That would be up to a judge. There is simply a legal requirement that they provide the source. Until they do so, they are violating Marlin's GPL.

It is the same with any contract. A promise to do something _eventually_ does not satisfy a legal requirement to do that thing, until that thing is actually done. That is one reason suits get filed.

> And then consider that there maybe proprietary comments, code that relies on proprietary functions or modules, and generally cleaning the code up to meet their standards.

The requirement from GPL is that the source code for the binary must be released, at the same time as the binary, and for the exact version of the binary as it was released. This also applies to all historical versions. The source code versions released must be in a 1:1 correspondence with the binary versions released. There is no allowance in GPL for removing proprietary comments or cleaning up code to meet their standards after releasing a binary. Once a binary has been released, it is a requirement to release the source code that was used to compile it. See https://gpl-violations.org/faq/vendor-faq/.

> If you want to see it faster, then give them constructive reasons that it is worth putting more man power on that project.

It is not a question of how Prusa should allocate its manpower. That is up to Prusa. They are legally obliged to release the source code, period. Obviously there are legal, ethical, and community relations reasons why it would be to Prusa's advantage to come into compliance, but it is not my job to define those advantages for them.

I will say that I personally find it disingenuous that a company like Prusa profits commercially from the IP provided _for free_ by the Marlin authors, yet fails to comply with the terms of Marlin's license agreement. They claim to support open source, yet they deliberately added a mechanism to the Buddy board which makes it difficult for owners to install their own firmware - "the appendix". Then they added code to the free software that checks the appendix and acts as a gatekeeper to modifying and improving the software itself, but they are withholding the source code for this tightly integrated additional piece. The entire spirit of the GPL - and free software - is to allow anyone to make their own changes to the software.

SamuelHaleMN, 2022-05-04:
So... Nearly a year later and still no resolution on this?

JustAnother1, 2022-05-15:
@dragomirecky you said:

> But hey, that being said, we are planning to open-source the bootloader slightly_smiling_face. Even though it is not a current priority, we are taking steps towards making it eventually happen.

If it is purely a matter of developers are busy and this is a low priority then let us help!

If whatever needs to be done to get the boot loader to be able to be published can be outsourced then I would be willing to do that. I could sign an NDA, get the source and the todos. I would do all the todos. And then you can publish the result as open source. And I would do all that for free! What more could you possibly ask for?

The XL is out now. Do we also need to wait for the mk4 to be out before the devs can sign off on releasing the boot loader?

Jookia, 2023-04-20:
The MK4 is out and also uses the same proprietary bootloader.

Prusa-Support, 2023-06-10:
Thank you for writing to us.

Open-sourcing the boot loader is not due nor promised, but it was in our plans.
Our plans about this haven't changed but the boot loader has been deeply reworked in the last few years and I believe it is still being worked on.

Please be advised, this may take some time but I'll send a reminder to our developers right away.

On a side note, as our team keeps on expanding so outsourcing may not be necessary.
The firmware is open source though and we can take into consideration Pull Requests from volunteering collaborators. There are several open issues and PRs out there waiting for help. 🙂 

Michele Moramarco
Prusa Research

JustAnother1, 2023-06-10:
> Open-sourcing the boot loader is not due nor promised, but it was in our plans. Our plans about this haven't changed but the boot loader has been deeply reworked in the last few years and I believe it is still being worked on.

Here https://blog.prusa3d.com/the-state-of-open-source-in-3d-printing-in-2023_76659/ @josefprusa said and I quote:

"Our desktop 3D printers will always be open source."

"The standard GNU GPL license under which our printers and software are available..."

I think that the bootloader of the Mini+ is part of the Mini+. And that is a  desktop 3d printer,.. I think you see where I'm going with this. The bootloader is also "software".

And yes I read that blog post and understand the reasoning, and the issues that arise from cheap Chinese clones.

I think It would hep in this situation to split the problem in smaller and therefore probably easier to solve parts. I think an important difference is software vs hardware. Open source software (especially firmware) is something very different to open source hardware.

Right now Prusa is not doing open source firmware! The STM32 is a very popular chip and there are many open source bootloaders available for it. Firmware is also very specifically designed to fit the hardware. The bootloader also includes the signature check for Prusa signed firmware. Therefore nobody who would want to do an 1:1: copy of the printer would benefit from the source code for the boot loader.

Who could benefit is the community that wants to look into it and wants to experiment with it. Also Prusa could benefit from it if the community finds bugs or sends a pull request for a new feature. Not merging pull request because they are of bad quality is a sign that software quality is taken seriously! Feel free to use this to help the community to produce better quality firmware. We are in this together!

That the bootloader has been worked on and is still being worked on is a reason to make it open source, not to hide it away. Once the development on it has stopped the community can not help any more. Also knowing how older versions worked could be helpful. Asking for help when bad actors violate licenses and make the open source unworkable is OK. But making it impossible for the community to help and then complaining that the community does not help is just stupid.

If you want the open source business case to work than it wont happen with a new license that makes all the problems go away. You rather have to look at all the issues separately and try to find solutions.

If I only look at the boot loader issue presented in this issue then I'm not proud on Prusa. And that is a sad situation. In this small issue Prusa does some of what the blog post accused others to do. If that is how Prusa wants to save open source, then it will not work, and that would be a very bad outcome for us all.

Prusa-Support, 2023-07-04:
> "Our desktop 3D printers will always be open source."
>
> "The standard GNU GPL license under which our printers and software are available..."
>
> I think that the bootloader of the Mini+ is part of the [...]

Please see https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1440#issuecomment-861301974.

> Right now Prusa is not doing open source firmware! The STM32 is a very popular chip and there are many open source bootloaders available for it. Firmware is also very specifically designed to fit [...]

Yet we are planning to open source it.

> If you want the open source business case to work than it wont happen with a new license that makes all the problems go away. You rather have to look at all the issues separately and try to find [...]

Your feedback is most valuable to us.
Please visit our [forum](https://forum.prusa3d.com/) for open discussions and feel free to share your thoughts as a comment in the relevant [blog post](https://blog.prusa3d.com/the-state-of-open-source-in-3d-printing-in-2023_76659/).

This request will remain open on GitHub for the time being.

Michele Moramarco
Prusa Research

JustAnother1, 2023-07-04:
> > "Our desktop 3D printers will always be open source."
> > "The standard GNU GPL license under which our printers and software are available..."
> > I think that the bootloader of the Mini+ is part of the Mini+. And that is a desktop 3d printer,.. I think you see where I'm going with this. The bootloader is also "software".
> 
> Please see [#1440 (comment)](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1440#issuecomment-861301974).

I did not argue that the bootloader would be part of the "Prusa-Firmware-Buddy". I argued that the bootloader is part of the printer and Josef claims his printer would be open source.

> Yet we are planning to open source it.

Once you have open sourced it then you can claim that prusa printers are open source. Until that time they are not.

> Your feedback is most valuable to us. Please visit our [forum](https://forum.prusa3d.com/) for open discussions and feel free to share your thoughts as a comment in the relevant [blog [...]

Been there, done that, ... ( even with the same name as here, you could have found that)

I don't understand the reason why you are not just pushing whatever code is currently available. The release of the MK4 and the XL is done now. So no more secrets that could be reveled by releasing the source. I also suggested to help (under NDA) to get the source into a releasable state. But that offer was declined.
So it could only be an issue of politics, or should I say waiting for the decision on the new license that Josef mentioned in the linked blog article?  "politics" comes from "poly" meaning many and "ticks"  ;-)

I can only recommend that in that discussion you clearly separate between open source software and open source hardware. As the arguments and options to deal with people ignoring the license are very different. Also the damage caused by violators are different. 

If Prusa has to deviate from being completely open source then deviating only on the hardware or only on the software side is better than deviating more in a general form.

From a cost / benefit analysis I can imagine that there is information that, if not open sourced, would harm the 1:1 clones more than it would harm the community. But the bootloader is not such an information.

If you need more feedback from me my email is my profile.

PS: Sorry for being so direct. But I don't like this very polite way of saying basically nothing. And I don't wand Prusa to go closed source.

Jookia, 2023-07-13:
I just want to note here that Marlin and the Prusa firmware is specifically licensed under the GPLv3 which REQUIRES Prusa not just release the source code of Marlin but provide a way to install modified versions of Marlin. The bootloader does not allow this and as such violates the anti-Tivoization clause of the GPLv3.

This is especially ironic considering Prusa has spent the last few months complaining about companies not respecting their licensing.

theacodes, 2024-05-21:
This issue is still relevant and should not be automatically closed.

JustAnother1, 2024-07-22:
Hey @Prusa-Support you said:
> This request will remain open on GitHub for the time being.

Just post the link to the sources, then you can close the issue. Doesn't hurt if it takes less than 60 days ;-)

Regarding the discussion about abandoning open source, new license and Chinese clones:
Do you really think that what they need is the Bootloader code?

theacodes, 2024-07-23:
This issue is still relevant.

In a private conversation with Josef, he indicated that he had some concerns around releasing the source to the bootloader since it may be abused my malicious users. I'm re-affirming my offer to assist in resolving those concerns and taking the time to prep the code for public release.

JustAnother1, 2024-07-23:
> I'm re-affirming my offer to assist in resolving those concerns and taking the time to prep the code for public release.

I would also be willing to put in work to get the code ready for release.

Jookia, 2024-08-12:
It's been a year since I last dropped by. To my knowledge this issue now affects 4 printers: The mini, the mk4, the mk4s and xl. The Prusa website still labels the printers as 100% open source and links to this repository for source code, which isn't available. The public and customers believe these printers are hackable, open source hardware and firmware. People are paying a lot of money for printers that are suppose to be fully open source, and hackable. Currently Prusa is delivering neither. That's not good.

Prusa being so tight lipped about the code being used by bad actors certainly hints that there's some security by obscurity happening. Perhaps it's time to start reverse engineering this bootloader and audit it?

theacodes, 2024-10-14:
this issue is still relevant

JustAnother1, 2024-10-25:
Just stumbled over this:

A reason for arguing that the boot loader must be open source is that it shares a common address space with the firmware.

The "communication" between the firmware and the boot loader also works over this shared address space.

JustAnother1, 2024-12-23:
> Hey @matthewlloyd and @JustAnother1,
>
> Thank you for your patience regarding this issue. To answer the primary question here: the bootloader is a separate project from the Buddy firmware. The Buddy [...]

That is not true anymore. The current firmware(6.1.2) comes (according to your web site) with the bootloader(2.3.5) included. So same binary, definietly part of the Buddy firmware, and you distribute it as part of the BBF file.

So according to **your** argumentation you are now obliged to release the bootloader!

> But hey, that being said, we are planning to open-source the bootloader 🙂. Even though it is not a current priority, we are taking steps towards making it eventually happen.

That comment is from 2021. "eventually" has not happened yet. Long after that comment @josefprusa  made this emotional blog post about those evil companies that do not comply with the licenses and therefore destroying the open source community,...

Can't you just come clean? Either you are open source, then close this issue with the link to the bootloader code, or just admit that you are not open source anymore and close this issue with that.

Otherwise I would start to wonder what evil stuff you are hiding in the bootloader,...

ekholm, 2025-01-17:
> All they have stated is that "we are planning to open-source the bootloader 🙂. Even though it is not a current priority, we are taking steps towards making it eventually happen."
> 
> While I applaud Prusa's agreement to come into compliance, their statement is not sufficient under GPLv3. Since the bootloader is dynamically linked, they are _legally obliged_ to provide the source code, and furthermore, it needs to be provided _in a reasonable period of time_. It has already been _two months_ since I opened this request. "Eventually" is not good enough. It is not a question of priorities - it is a legal requirement.
> 
> When I have made similar requests to other companies, they have been handled within _days_. For example, I asked BigTreeTech to provide the source code to their modified version of RRF for their Octopus board, they did so in _two days_ ([bigtreetech/BIGTREETECH-OCTOPUS-V1.0#40](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/issues/40)). When I asked Fokoos to provide the source code to their modified version of Marlin for their Odin 5 F3 printer, they supplied it within a _few hours_.

I guess you need to read more about how a bootloader work in embedded systems, and what dynamic linking is. 
Its quite a far stretch to say that the boöotloader and firmware is dynamically linked. It's like saying that an OS is dynamically linked with any (pre) installed applications. Which is obviously wrong. Also, shared memory communication does not imply that two pecies of software is tied together as one.

When flashing a prestine board it is convenient to pack the bootloader, firmware and some meta data in the same blob. It's a PITA to flash those three separate binaries as indivildual items. 
The bootloader need not know anything about the FW or its internal working. Both the FW and the bootloader are self contained (in most cases). 
What the bootloader do need to know is the presence of a FW or not. That is, should the bootloader start a FW or not. This can be done in a mulltitude of ways, typically involving checksum of the flash content and comparing it to a known value. In bootloaders I have been involved in developing, we have the checksum stored as the last word in flash, or as an alternativ in the meta data region.
The bootloader also need to know the entry point of the firmware, the first instruction to execute in the firmware. Depending on architecture (ARM, AVR etc) it is placed at a well defined offset after the bootloader.

The bootloader does not do any linking prior handing over the execution to the firmware, and the firmware does not call any funtions in the bootloader. In this respect the bootloader and the firmware is two separate entities placed after each other in memory. In otherwords, an aggregate which is conveniently distributed in the same container/media. 
There is no bi-directional communcation taking place between the bootloader and the firmware.

A bootloader is in most cases a very simple piece of software. I don't know about the Prusa bootloader, but if it have any kind of networking capability for downloading new FW they I guess they would like to have signing keys or tokens embedded in the bootloader. Which for obvious reasons need to be "closed source".

Jookia, 2025-01-17:
1. Signing keys and tokens don't require source code to be closed, they use public key cryptography. Only the public key is in the bootloader. If they are putting other secrets in the bootloader, then they've made a mistake. Firmware that is closed source that uses symmetric keys is easily broken, look at the Anycubic Kobra 3. The visibility of source code doesn't really matter.

2. Marlin's license requires Prusa to let you install a modified version of Marlin on your device if you use it. The GPLv3 is clear that you cannot lock down devices like this if you want to use its software.

Prusa doesn't have to use Marlin's code, but they've chosen to and have to abide by the license. Or just ignore it I guess? More importantly:

3. Prusa says their printer  bootloader is open source. Their website advertises their firmware and printer as open source. Their customers bought a printer saying their code is open source.

If Prusa doesn't want to release their bootloader code, that sucks. They should close the issue and state their position publicly. But right now they are intentionally leading people on by saying it will be open source and leaving this issue open as something that needs resolving. They have been doing this for years even though they had plenty of time to figure this out before even releasing the Mini and newer printers. They are a company, they have a legal department, they have people here and likely internally who are offering to help get the code released.

CZDanol, 2025-01-17:
I'm sorry, I cannot comment on the bootloader release, but I would like to point out that you can:

1. Flash anything you want on the printer without having the bootloader source (you just need to break the appendix, which doesn't even void warranty in general anymore btw - though if your modified firmware causes some damage, that is ofc on you)
2. You can even use your printer without the bootloader altogether perfectly fine (you can build our firmware in the "noboot" variant and flash it through the jtag interface - but then you will have to keep using it for flashing new fw, or flash the bootloader back)

ekholm, 2025-01-17:
Is it not possible to install Marlin FW on the Prusa hardware without the bootloader? I'm sure they let you install whatever FW you want (of course voiding any warranty). At leats on the XL it's possible.

Regarding signing keys, I don't know about the Anycubic issue. 
If the bootloader is closed or open source it will not make any difference in how hard it is to expose the stored keys. A properly made bootloader that also enforces the correct microcontroller fuse bits will make it *very hard* to break. 
Yes, with the public key cryptography you only need one of the keys, which one depens on use case. But the scenario where web-server need to know that it's an "original Prusa" then the printer need a secret key...

If Prisa says that the bootloader is indeed open source or based on GPL source then they should of cource make it publicly accessible. And yes, it's a problem missleading customer into beleiving that everything about Prusa is open source.

Jookia, 2025-01-17:
>     1. Flash anything you want on the printer without having the bootloader source (you just need to break the appendix, which doesn't even void warranty in general anymore btw - though if your modified firmware causes some damage, that is ofc on you). The bootloader will then let you flash any firmware you provide.
> 
>     2. You can even use your printer without the bootloader altogether perfectly fine (you can build our firmware in the "noboot" variant and flash it through the jtag interface - but then you will have to keep using it for flashing new fw, or flash the bootloader back)
> 
> 
> So we are NOT locking your printer.

This is a little misleading. Prusa has a clear benefit here of being the sole authority for what runs on unmodified printers. This is enforced using a lock that denies non-Prusa firmware. Users aren't given the option to control this lock or bypass it, instead Prusa says you must destroy the lock. Users do not ever get to experience the benefits of the lock that Prusa does, namely signed images for secure boot. The printer's functionality is degraded and the asymmetry is still there: You get to pick to run only Prusa software, or any random software someone plugs in.

CZDanol, 2025-01-17:
Well having an "unmodified printer" implies running an "unmodified firmware" on it in my eyes - so that sounds perfectly reasonable.

Technically, you can write your own bootloader that checks for any key you want and that disregards that the appendix is broken and still requires the firmware to be signed. Yes, it would be less convenient as you would have to write the bootloader from scratch at this moment, but you are still not functionally limited.

ekholm, 2025-01-17:
If you know your way around hardware there is no problem reversing the "breaking of appendix". PCB vias are available so it's possible to patch back the function of the appendix. Granted that you can't reverse it in a way that's not vissible. 
Best I think would be to be able to break the appendix in a permanent and visible manner and then be able to add jumpers to revert it's functionality. 

I guess from a warranty point of view it make sense that any alterations are permanently *visiible* even if their effect can be undone. You would not like to end up in a situation that a custom FW destroy some mechanical part and the user then revert the changes and blame the company. 

I completely agree that a product should not be useless after a vendor decides to not support it anymore. Like your old iPhone or similar. *But* with Prusa having the firmware open source, you can still use and upgrade your printer even after Prusa has given up on it. The point is, you don't need the bootloader to use the printer, it's the firmware that is important and the hardware specs (Schema). If you for some reason *need* a bootloader for your printer, there are plenty available that are open source.

ekholm, 2025-01-17:
Looking at the Buddy 1.0 PCB, it is possible to add jumpers to revert the appendix destruction.

JustAnother1, 2025-01-21:
@ekholm The micro controller used is an ARM Cortex-M. These chips have one time programmable (OTP) Memory. And ARM really likes their TrustZone stuff. (The one in the RP2350 was recently broken(38c3 had a talk about it) because Raspberry Pi f'ed up)

So there are better ways to do this then keeping the boot loader a secret.

Back in the days when the Mini bootloader was a separate thing the firmware files were signed, but not encrypted. Did they change that?

If they are still not encrypted and the bootloader contains some kind of secret key or something then it is in the update file in plaintext. Would be a stupid beginners mistake to do it that way.

Has anybody tried doing a disassembly of the bootloader?  Or knows how to split the update file into bootloader and firmware parts?

CZDanol, 2025-01-21:
The bootloader is not encrypted and we are not doing any secret keys or anything like that there :) You will only find our FW signing public key in the bootloader.

If you look at https://github.com/prusa3d/Prusa-Firmware-Buddy-Private/blob/bf217dde40f74895430a064e307739521b2a3f16/utils/bootstrap.py#L74, you can see links for the bootloader binaries for all the printers - those are the binaries that are used to update the bootloader.

ekholm, 2025-01-21:
@CZDanol you mean, i guess?
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/utils/bootstrap.py

the link you provided is "private" it seems

CZDanol, 2025-01-21:
Yes, I meant https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/136fb25a9f97434ea47b6c67b4d69eb66f83469a/utils/bootstrap.py#L74, sorry

JustAnother1, 2025-01-21:
> The bootloader is not encrypted and we are not doing any secret keys or anything like that there :) You will only find our FW signing public key there.

Then why don't you release the source code then?

CZDanol, 2025-01-22:
I cannot say at this moment, sorry.

ekholm, 2025-01-22:
Why is the source code for the bootloader such a big deal? It's a simple piece of software and obviously doing nothing magic with the printer. 

Speculating, there are a plethora of reasons why Prusa at the moment can't (or would not like) to make the source public. For instance, all comments are written in Czech, or symbol names might not be "suitable" for the public eyes or simply lack of resources/priority issue. It's no point releasing the software to the public if it not has been cleaned up and reviewed, otherwise some one would start whine about the code look like garbage or is incomprehensible.

Ask your self, would you like Prusa to spend time on making the bootloader public or spend time acting on the bug reports on the actual printer FW. My stance is clear, bootloader seems rock solid, but the printer FW do need some extra attention. I do understand the idea behind free and open source software. But on this case, the software in question is not critical for the function of the hardware and does not restrict its usability.

To make it clear, my sole relation with Prusa is as a customer and I do hate closed source. Even though my bread and butter work is with closed source on embedded devices (like bootloaders...).

JustAnother1, 2025-01-24:
@ekholm I agree the source code for the bootloader is not a big deal. The bigger deal is why prusa thinks they need to keep it a secret.

This issue was opened in 2021. Therefore being busy right now can not be the reason for not open sourcing the code.

If they can release the firmware code and it is not in Czech or otherwise unrepresentative, then why should the bootloader code be.

Open sourcing the bootloader takes less them 5 minutes. They have the code all they need to do is a git push.

But you are also wrong. The bootloader is critical for the function of the hardware and it restricts its usability.

Also @CZDanol message implies they have a valid reason that is a secret.

For people that want to take a look: here is the disassembly of the mini 2.3.5 bootloader:

[bootloader.s.zip](https://github.com/user-attachments/files/18538818/bootloader.s.zip)

JustAnother1, 2025-01-30:
Using `strings bootloader.bin` on the file that @CZDanol linked you can find these strings:

`Unable to verify
the extruder type,
check the wiring
and connectors.`

That is a bit scary. It sounds like they now also want to somehow link the extruders to the hardware ( Error messages for the wrong version of buddy board or the wrong printer are also in there). Or maybe future Extruders need signed Firmware updates?

`Erasing XFLASH & EEPROM`

I guess that "XFLASH" is an external Flash. Did not know they needed/used that.

`assertion "%s" failed: file "%s", line %d%s%s`

Hey they use assertions in production code. Good to know.

`/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/src/newlib/newlib/libc/stdlib/mprec.c`

And they use Jenkins on a Linux (Mac?) system. With gcc 10. And compiled the bootloader in 2021on October 18th.  I hope they do more than a build server, but have many unit tests running there also. Do not know why they need to flash this information to every-bodies printers though.

Using `hexdump -C -v  bootloader.bin` shows that there is a lot of empty space in the file. So maybe they are ashamed of their bootloader code?

Also this bootloader seems to be Mini specific. In the old version, we looked at, the name of the XL leaked. In this there is no reference to the XL. But this extruder type check error message,...

danopernis, 2025-01-30:
> Unable to verify the extruder type, check the wiring and connectors.

This is nothing scary or secret, it is an error [publicly documented](https://help.prusa3d.com/article/extruder-not-detected-20615-mk4-mk4s_813941) in our knowledge base.

Now, the particular reason why bootloader needs to communicate with loveboard is to distinguish between MK3.5 and MK3.9/MK4 in order to disallow flashing mismatching firmware. Those printers share the same hardware (xbuddy board). Arguably, this string should not be present in the bootloader, but since all our bootloaders share the same codebase and we don't have LTO enabled yet, it is present in the binary.

> I guess that "XFLASH" is an external Flash. Did not know they needed/used that.

XFLASH is indeed external flash which, as documented on our [opensource page](https://www.prusa3d.com/page/open-source-at-prusa-research_236812/) is indeed present on all our boards. STM32F4 platform only contains 2MB of FLASH (1MB of FLASH in case of MINI) so without XFLASH, there won't be any space for fonts, images, firmware for ESP Wi-Fi add-on, daughterboards on XL etc. All of this is public information and could be inferred directly by reading the documentation or the firmware source code.

> /mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/src/newlib/newlib/libc/stdlib/mprec.c

We are linking again standard arm toolchain, like we do in the firmware. Strings you are seeing come from their buildsystem and their build of `newlib`.

> Using hexdump -C -v  bootloader.bin shows that there is a lot of empty space in the file. So maybe they are ashamed of their bootloader code?

I don't understand why "empty space in the file" would be considered something to be ashamed of. That being said, there are some aspects of the codebase, mainly related to coding style, which could be improved.

> Open sourcing the bootloader takes less them 5 minutes. They have the code all they need to do is a git push.

Reiterating on what was already told, opening the bootloader is not a decision I am allowed to make, although I would like it much more then answering questions like this on github 😫

theacodes, 2025-01-30:
Just dropping in to say that my offer to assist with open-sourcing the bootloader still stands- I'm the president of the Open Source Hardware Association and highly experienced in helping organizations take proprietary code through the process of publishing it under open source licenses. I'd be happy to listen to the concerns of Prusa's stakeholders and present a publication plan that addresses those as well as assist in any modifications needed to get the codebase into a publishable state.

JustAnother1, 2025-02-01:
I looked into the disassembly and I now think I understand why open sourcing is such a big issue.

They have probably used some kind of Tool / Framework to write the bootloader. It is not just a pure C project like the Firmware.

If you look into the bootloader.bin file (or my posted disassembly) you will see that the it starts out with what seems to be a Vector Table followed by normal code. But that ends at address 0x30f. (that is very little code) Then a lot of 0x00 until 0x4000.(a big gap) At address 0x13160 we have 15 Strings that are fixed length of 150 bytes. With the 15th being strange. In other places we have the normal C style strings (0 terminated). Starting at address 0x1d160 for example.

In these sections there is something that should be normal code but is not only code:
0x4000 - 0x130bf
0x13ad0 - 0x1d169
0x13eda - 0x1dee6

Whatever is in these sections only partly can be de-compiled(illegal registers used or unknown instructions). It probably has some data mixed in or is encrypted.

At the very end there is a CRC and "2.3.5"(at address 0x1fffa) as ASCII. And that is basically it, the rest is 0x00.

So they most probably used some commercial proprietary tool that generated this mess. And that is the reason they can not open source it.

They probably thought that they could not do it themselves (with the crypto and not wanting to update the booloader all the time) and therefore went that route. Probably made financial sense to pay the tool vendor instead of programming it themselves. 

And maybe we should have a discussion if open source vendors should be allowed to do such a thing. But to kick of that discussion Prusa would need to confess to doing it. Without that we just have the missing bootloader code as proof that Prusa printers are not open source, case closed!

@danopernis 
>  STM32F4 platform only contains 2MB of FLASH (1MB of FLASH in case of MINI) so without XFLASH, there won't be any space for 

but also

> I don't understand why "empty space in the file" would be considered something to be ashamed of. 

Maybe less empty space would make more functionality fit? And no XFLASH or a smaller XFLASh would make the board cheaper and would therefore allow for a cheaper or better printer.

danopernis, 2025-02-03:
@ekholm @theacodes I will bring this to internal meetings with some folks with more power/responsibility, but can't promise anything. I am sure you understand how difficult those tasks can be. Your willingness to help is much appreciated, at least by me personally.

@JustAnother1 sorry I am not reacting to you anymore. I wanted to provide some information for you in good faith but clearly I am just wasting my time here. You clearly lack expertise in embedded software engineering and reverse engineering, as any expert in the field would confirm. Being surprised about parts of the binary image containing data, not only executable code (as already told, there are fonts and keys to verify firmware signature in the firmware), expressing concerns about "empty space" (yes, zeroes may have their meaning even when they look empty to you), being surprised that not all strings are zero-terminated, making wild theories about "generated code" by some mysterious "tool vendor"...

I am willing to answer any meaningful questions, but this is just wasting time unfortunately.

To reiterate, bootloader is not open source at the moment, firmware is open source. Bootloader code is not required for modifying the firmware. Either you can remove "appendix" and let the bootloader flash your unsigned modified firmware, or you can build the firmware in the "noboot" preset and overwrite the entire FLASH. The sole task of the bootloader is to prevent flashing unsigned firmware. If you want to, you can lift the hood and do whatever you want to the printer.

theacodes, 2025-02-03:
> I will bring this to internal meetings with some folks with more power/responsibility, but can't promise anything. I am sure you understand how difficult those tasks can be. Your willingness to help is much appreciated, at least by me personally.

Thank you! I'd spoken to Josef about it previously, but it's been quite a while and I think I remember him dealing with a new child at the time, so no worries.

ekholm, 2025-02-04:
Hi,
you do quite a lot of speculation....

I would be suprised if Prusa is using a commercial tool to create the bootloader. It's such a simple piece of software.
One thing that you absolutely want to avoid is to run out of space of the bootloader section. Unless you are really tight in space you would want some gap in between the bootloader and application code. So that you can upgrade the bootloader with a larger one if need be down the road. 

It's not strange to have "unknown instructions" embedded in the code. A normal switch statement in C might end up as a jump table in the assembly. 

Some speculation of my own, maybe they have sections in the bootloader section containing some minmal code and data to show on the display if external flash for some reason is not available. Then you might divide the bootloader section into chuncks continaing bootloader code, graphical data, etc Which would inevitable be some gaps between those sections.
Another reason for big gaps would be to have the possibility to install a new bootloader whilst preserving the old one if installation fails.

I don't know how large the external flash is, but firmware for the other boards as well as language translations, and graphics takes a lot of space. So even if the bootloader was not present at all, I doubt that the exernal flash could have been removed.

Just because you don't understand the layout or the code it does not mean that Prusa is doing some anything malicious.

br

> I looked into the disassembly and I now think I understand why open sourcing is such a big issue.
>
> They have probably used some kind of Tool / Framework to write the bootloader. It is not just a [...]

[2 more replies]