prusa3d/Prusa-Firmware-Buddy #1444: [QUESTION]Why the 100℃ default bed temp for flex filament.

RAMilewski, 2021-05-30:
The Prusa Material Guide recommends a bed temperature of 50℃ for all flex filaments except for ColorFab nGen Flex (85℃).  So why does the firmware set the bed temp to 100℃ when changing filament or doing a first layer test.   In my case this makes the textured steel sheet extremely difficult to clean.

DRracer, 2021-07-30:
@RAMilewski this is very suspicious - in the code we have 240C for the heater and 50C for the bed. Are you sure you selected the correct preheat preset?

JohnnyDeer, 2021-08-10:
@RAMilewski, tested on 4.3.1 and 4.3.2. In all cases flex preheat to 240/50°C. Please try it again and make sure if your printer really heatbed to 100°C, thank you.

JohnnyDeer, 2021-09-02:
This issue will be closed soon if autor will not respond.

closed 2021-09-06 by JohnnyDeer