prusa3d/Prusa-Firmware-Buddy #1448: [BUG] Running an unload through M702 causes the filament runout sensor to trigger M600 on Octoprint

Jonas-Mikkelsen, 2021-05-31:
**Printer type** - [MINI+]

**Printer firmware version**-  4.3.2-RC1

**Original or Custom firmware** Original latest release

**Optional upgrades** - Filament runout sensor

**USB drive or USB/Octoprint**
Connected via micro usb to a Raspberry Pi 4 4gb running octoprint.

**Describe the bug**
Inputting M702 in the terminal runs the unload procedure as expected however when the filament exits the filament runout sensor it triggers the newly added serial filament runout behaviour.

**How to reproduce**
When connected via octoprint unload the filament via the command M702.

**Expected behavior**
  The filament unloads as normal and the printer recognizes this.

zoltan-l, 2021-06-02:
Hi Jonas, 
did it happen during print or in the printer main screen mode? 
The terminal was a octoprint terminal?

Jonas-Mikkelsen, 2021-06-02:
This happened on the octoprint dialog page, however it was not on the main screen and this was not during a print.

JohnnyDeer, 2021-06-02:
I can confirm this issue. @Jonas-Mikkelsen thank you for reporting.

coaxial, 2022-07-19:
Still experiencing this issue on firmware 4.3.4.

Prusa-Support, 2022-07-19:
Hello! Thank you for your feedback, sorry to hear that you still have the issue with 4.3.4. I'll bring it to our devs attention to further review it. In case of anything more, I'll update on this thread, in case anyone has a similar issue or in case of more feedback, do not hesitate to let me know by adding more comments. 

Alessandro Pantaleo
Prusa Research

VreeDogNight, 2023-04-02:
Prusa Firmware: 4.4.1+4194
OctoPrint: 1.8.7
This is still happening with the current firmware. Would it be possible to implement `M405: Filament Sensor on` and `M406: Filament Sensor off` to manually be able to fix this issue if it is not possible to fix this bug with `M702`?

Prusa-Support, 2023-06-07:
Thanks for reaching us.

In a summary, the filament sensor doesn't detect that the filament has been pulled out during a color change; and the suggested solution is to disable the filament sensor via codes. 
Is this accurate?

A user in issue #2970 also reported that the same happens when the printer is connected via Prusa Link (not only Octoprint) and if the print is run via LCD menu, but reinserting and pulling out filament a few times may unfreeze the filament detection.
We would appreciate a confirmation.

Michele Moramarco
Prusa Research

VreeDogNight, 2023-06-08:
> In a summary, the filament sensor doesn't detect that the filament has been pulled out during a color change; and the suggested solution is to disable the filament sensor via codes. Is this accurate?

No that is not an accurate summary. The issue is that when you send a M702 in the terminal it runs the unload procedure as expected, but when you pull the filament out of the filament runout sensor it triggers the M600 change filament process. I was wondering if it would be possible to update the firmware to stop that from happening or at least add in the option of turning the filament sensor off and on again with the `M405: Filament Sensor on` and `M406: Filament Sensor off` commands.

Prusa-Support, 2023-08-25:
Alright, thanks for clarifying.

Our developers should briefly overhaul Octoprint-related issues in the near future and possibly resolve this undesired filament runout detection.

Implementing the codes `M405` and `M406` may not only be an alternative solution for this issue but also sounds like a fair feature request on its own. It may deserve a separate issue.

Michele Moramarco
Prusa Research

coaxial, 2023-08-26:
It's good to finally read some commitment to making Octoprint work with the Mini/MK4.

closed 2024-06-17 by github-actions[bot]