prusa3d/Prusa-Firmware-Buddy #1452: [FEATURE REQUEST] Option to edit steel sheet profiles

JustAnother1, 2021-06-03:
**Printer type** - [MINI]

**Is your feature request related to an existing problem? Please describe.**

After levelling the steel sheet on prints I detected elephants foot. I therefore suspect that I levelled the nozzle to close to the sheet. I just want to increase the distance between build plate and nozzle a bit. But I can not do that without running(and printing) the first layer calibration wizard thing.

**Describe the expected functionality**
I want to select the profile of the sheet and change the Nozzle offset in the menu without the 1st layer calibration print.

Jonas-Mikkelsen, 2021-06-05:
You should be able to change the live-z value under calibrate and it should save that to the profile.

Works like that in my experience!

JustAnother1, 2021-06-05:
@Jonas-Mikkelsen Thank you for the hint. That was not obvious to me.

It would still be nicer to have this option also available in the sheet menu. That would allow to check in on the current value and change values without the need to select the sheet first.

JohnnyDeer, 2021-06-07:
Hi @JustAnother1, as @Jonas-Mikkelsen wrote you can do that in calibration menu. 
Your suggestion in comment is already existing issue #1445. 
I'm closing this issue as invalid.

closed 2021-06-07 by JohnnyDeer