prusa3d/Prusa-Firmware-Buddy #1454: [BUG] Extreme stuttering when resuming during bed leveling process

GammaGames, 2021-06-05:
**Printer type** - Mini

**Printer firmware version**-  4.3.1

**Original or Custom firmware** Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint** - USB

**Describe the bug**
  When resuming a print paused during auto bed leveling, the head goes in reverse on the final row before reversing the final row and rattling the printer violently. It shook by ~45 degrees on the table.

**How to reproduce**
  1. Start a print
  2. Pause the print just after the penultimate prod
  3. Resume the print
  4. Watch in horror as the head goes to the end of the arm, *something* happens, and the printer rattles the table

**Expected behavior**
  It should do the final leveling point

**G-code**
  I've been able to reproduce this on a few PETG prints from PrusaSlicer using the Prusament profile. It probably does it for PLA as well, but I'm too scared to test

GammaGames, 2021-06-05:
Seems slightly different than #981, the nozzle was nowhere near the bed when I resumed.

JohnnyDeer, 2021-08-10:
Hi @GammaGames, I tested this issue on 4.3.2 and bug don't occurs. Could you please try it too?

GammaGames, 2021-08-10:
Will do tonight!

GammaGames, 2021-08-11:
I tried to recreate it and couldn't, thanks for checking back!

closed 2021-08-11 by GammaGames