prusa3d/Prusa-Firmware-Buddy #1456: [ENHANCEMENT] Make E-Steps adjustable

fhb, 2021-06-07:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

Enter what type of printer or upgrade the enhancement applies to.
**Prusa Mini +** - [MINI]
**Bondtech Dual Drive Extruder** - [e.g. Filament Runout Sensor]

**Is your suggested improvement related to an existing problem? Please describe.**
  A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the expected functionality**
  A clear and concise description of expected functionality.

matthewlloyd, 2021-07-06:
This feature is available in the unofficial Llama Mini firmware: https://github.com/matthewlloyd/Llama-Mini-Firmware#llama-mini

PippoAe, 2021-08-17:
Still waiting for this.

Dev01-D, 2021-09-07:
I believe this is now implimented in 4.3.3 RC! WOOP!

PippoAe, 2021-09-08:
URRMAGRD! That would be awesome! :)

JohnnyDeer, 2021-09-13:
Hi, as @Dev01-D wrote, this feature is in 4.3.3-RC. I´m closing this issue.

closed 2021-09-13 by JohnnyDeer

SunshineSupremo, 2025-08-14:
Greetings everyone; just in case anyone else has still been struggling to find the answers to this issue, I thought I would post this info, which was profoundly helpful to me in solving it for myself: I asked BondTech about the best way to get their extruders working with the MK3.5S, and they referred me to this link, and said to use 830 E-steps...

https://help.prusa3d.com/article/experimental-settings-mk4-s-mk3-9-s-mk3-5-s-xl-mini_264396

Some thoughts: the 830 number is supposed to be based on 16 microsteps, and also on the BondTech gearing being 3:1 compared to Prusa's 1:1... my printer was resetting its E-steps to 266 no matter how many ways I tried to M500 that $#!+, which would imply that perhaps my exact number is 798 (aka 266*3), but when I did the ol' 100mm extrusion test, I came out with 906 as my subjective requirement. All three of those numbers, in my case, seem to print just great (especially compared to whatever time-guzzling mess it was making before). Your mileage may vary and it is probably still worth fine-tuning accordingly, but this should get you started. Some forum users say the MK3.9/MK4 extruder motors are set at 32 microsteps, so possibly those need a number more like 415 for the E-steps, but I cannot vouch for that: do your own research as always, and make changes at your own risk ("...but what if the risk is that it starts working properly?")...

ALSO, in my case, probably simply due to the FW updates that have happened since Prusa posted that article, the menu mentioned in the above article was not "secret," it was just under Settings > Hardware, on the printer's control screen.

Also, there is a box there that just says "Make it work properly," which you should consider checking. 

Just kidding. But in the immortal words of Syd Barrett, what exactly IS a joke?

You're so close! Good luck!!!