prusa3d/Prusa-Firmware-Buddy #1457: [ENHANCEMENT]Satin Sheet Profile

chrishilbert, 2021-06-08:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MINI
**Optional upgrades** - Satin Sheet
**Is your suggested improvement related to an existing problem? Please describe.**
No

**Describe the expected functionality**
On the HW Settings screen for sheet profiles, please add Satin into the listening so I can have Smooth, Textured, Satin as my options.  This way I don't lose Textured1/2/3 and can keep the settings straight.

Dev01-D, 2021-08-02:
I would also like to see this added. I think as the sheet is now being widely used it would be nice to have the settings reflect the sheet in use.

Prusa-Support, 2022-05-18:
Hello! Sorry for intervening on this thread so late. I am closing this issue and marked it as solved, it has been indeed added in the meantime, in the 4.3.4 FW release. :) 

Alessandro Pantaleo
Prusa Research

closed 2022-05-18 by Prusa-Support