prusa3d/Prusa-Firmware-Buddy #1471: [BUG]No ip address when already printing

Fliens, 2021-06-15:
**Printer type** - MINI

**Printer firmware version**-  4.3.2-RC1

**Original or Custom firmware** Original

**USB drive or USB/Octoprint** USB

**Describe the bug**
I started a print and the lan cable wasn't plugged in yet. After pluggin it in it won't show an ipAddress in LAN-Info.
It also isn't available with the ipAddress it had before and there is also no time update when it is finished.

But the icon for LAN-Connection is there so it's a bit weird

**How to reproduce**
  Start print and then plug the LAN cable in

**Expected behavior**
  LAN-Connection Icon appears but printer get's no ip-Address

Fliens, 2021-06-16:
@JohnnyDeer I get an error that no static ipv4 address was set

Fliens, 2021-06-16:
It needed a full reboot of the printer

Is it no longer hot plugable?

OJFord, 2021-06-20:
I wasn't printing so wasn't an issue for me, but I also needed to reboot with 4.3.2-RC1. (Never tried on another version.)

closed 2024-06-17 by github-actions[bot]