prusa3d/Prusa-Firmware-Buddy #1474: [FEATURE REQUEST] Allow setting of time left metric

chris72205, 2021-06-16:
**Printer type** - Prusa Mini+
**Optional upgrades** - N/A

**Is your suggested improvement related to an existing problem? Please describe.**
No, this is not related to an existing problem.

**Describe the expected functionality**
I would like to be able to set whether the printer displays remaining time or when print will end based on a setting.  I have my printer connected to LAN but it's more straight forward (most of the time) for me to see how much longer is left instead of print will end at (and then doing the mental math to figure out how much longer is left).

Ideally when connected to LAN, it would show "Print will end" as it does now but also show remaining time next to the percent completed metric: `48% - 2h8m left`

reedy, 2021-06-20:
Dupe of #648?

chris72205, 2021-06-21:
> Dupe of #648?

Ah yeah, I tried searching before making a new issue but missed that one I guess.

closed 2021-06-21 by chris72205

reedy, 2021-06-21:
Search isn't the best on GitHub :)