prusa3d/Prusa-Firmware-Buddy #1480: [BUG] Changing filament mid print after running out seemed to change filament type

reedy, 2021-06-19:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI

**Printer firmware version** 4.3.1

**Original or Custom firmware** Original

**Optional upgrades** Filament Runout Sensor

**USB drive or USB/Octoprint** USB

**Describe the bug**
It seems that when the filament runs out mid print (as stopped by the runout sensor), the printer may seemingly change the filament type it thinks it is printed, even though it was not requested

**How to reproduce**
Start a long print with PETG loaded. Let the PETG runout, and the printer cool down because you ignore the alarm. Load some more PETG using the wizard onscreen (which doesn't ask which filament type), and then notice that it has reverted to PLA even though it wasn't told PLA was being loaded.

I'll have to play around  abit more and see if I can reproduce, and provide some pictures/vido.

**Expected behavior**
When reloading mid print, the printer should not (unless told to) change what filament type it thinks is loaded

**G-code**
N/A

**Crash dump file**
N/A

**Video**
N/A

murk-sy, 2021-06-20:
Duplicate of #1400

reedy, 2021-06-20:
Aha, thanks. I thought I'd seen something, but searching failed.

Closing as a dupe then!

closed 2021-06-20 by reedy