prusa3d/Prusa-Firmware-Buddy #1485: [ENHANCEMENT] LCD Info diplayed - ADD Layer Number printing out of the total number of layers

Martenz, 2021-06-24:
**Printer type** - [MINI]
**Optional upgrades** - [LCD info displayed]

Would it be possible to add some more info in the LCD of the print status?

I really like to use Octoprint but I have to say that I now prefer to use only Prusa LCD to tune a bit velocity temp etc while printing from there.. What I still miss is just where the print is, to compare for instance with the PrusaSlicer sliced model.. the percentage is fine but would also be nice to see the layer number that is printing out of the total number of layers so if something is weird I can go to the layer and see if it is a problem of settings or of the 3D model I can improve for better prints.

Is this a feature I can ask here?

Does it make sense also for someone else?

Thanks a lot,

Martino.

bdwheele, 2021-07-02:
That would be great to have -- often I'm printing something and there's a "gotcha" point in the print so I'll track where I am in PrusaSlicer and comparing it to what's coming out of the printer.

Without layer information I'm subtracting the z-axis height and doing a best guess.  This feature would take away the math, so I'm all for it.

Martenz, 2021-07-02:
Exactly I would like to have ot for the same reason!
Finger crossed.

jh-vo, 2021-08-06:
Yes, this is really needed function! Add it, please!

x10102, 2021-08-30:
Yes please, this would be so much better than estimating the layer from the Z-height.

yetisnack1, 2022-08-19:
+1 for this feature.  Besides being beneficial for troubleshooting, it's useful for manual color swaps, adding magnets, providing temporary supports (ie. sticking your finger under a long overhang ;) .  I find the mini's zlayer height useless due to it constantly changing and would much prefer to have the actual layer number displayed.

Prusa-Support, 2022-08-26:
Hello! Thank you for the request and for the comments, sorry for the late response from my side. Request noted and brought to our devs attention, in case of more info from our side we'll update in this thread. 

Alessandro Pantaleo
Prusa Research

CZDanol, 2024-05-21:
Hey, this is being considered as a part of internal ticket BFW-5403. Not promising it will pass through though. Also, not sure if it will work with Octoprint.

jim-roberts, 2025-03-19:
There has not been update on this open issue for over year. What is the status?

bkerler, 2025-08-25:
I wrote a PR for displaying the current layer in the footer display. The layer height is taken from the gcode meta (layer_height and first_layer_height variables that prusaslicer automatically adds). I do also take the z offset (if used) into account.