prusa3d/Prusa-Firmware-Buddy #1486: [FEATURE REQUEST] XY Skew correction

ericclinedinst, 2021-06-24:
Printer type- MINI

Can we please get xy skew compensation and an xyz calibration just like the mk3 and mk2 please? Seen lots of threads on prusa reddit, the prusa forum, and my own experience with having to manually manipulate the xy planes on the mini. The  calibration spots are on the bed, so why don't we have the feature enabled yet?

Thanks

matthewlloyd, 2021-07-06:
Llama Mini firmware enables 3-plane skew compensation: https://github.com/matthewlloyd/Llama-Mini-Firmware#llama-mini. It doesn't use the calibration spots, but you can just print the STL, measure with calipers, and use the supplied spreadsheet to calculate the correction factors.

ericclinedinst, 2021-07-06:
> Llama Mini firmware enables 3-plane skew compensation: https://github.com/matthewlloyd/Llama-Mini-Firmware#llama-mini. It doesn't use the calibration spots, but you can just print the STL, measure with calipers, and use the supplied spreadsheet to calculate the correction factors.

Yeah, that's exactly what I want to do. Cut the appendix and void my warranty. Thanks but no thanks. I think a lot of us would like to see xyz calibration in the initial configuration of the printer just like we had for mk2s and mk3.

krcm0209, 2021-07-07:
Duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/385

jmrp98, 2021-10-02:
I agree, I would really like to have XYZ calibration like the MK3 and MK2. The mini has a big problem with not being aligned and it is frustrating.

SiegelDaniel, 2022-03-02:
@Gaugeforever Can you supply a link to this mod? I cannot find it.

Gaugeforever, 2022-03-02:
Certainly, this is a link to my reply on the topic but you can navigate to the first page.
[Link](https://forum.prusaprinters.org/forum/user-mods-octoprint-enclosures-nozzles/prusa-mini-silicone-bed-leveling-mod/paged/15/#post-518551)

espr14, 2023-01-08:
See #385 for the possibility to set the skew compensation. Automatic calibration may come later.
Since the MINI has the lowest priority compared to MK3/XL/AFS etc. don't expect the skew compensation in original FW soon.

Prusa-Support, 2023-02-03:
Thanks for your input.
Same as https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/385, our developers could consider this feature request in future firmware developments.

Michele Moramarco
Prusa Research

closed 2024-06-17 by github-actions[bot]