prusa3d/Prusa-Firmware-Buddy #1492: [FEATURE REQUEST] Higher Density Mesh Bed Levelling for Prusa MINI+

imnotme, 2021-07-01:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - Prusa MINI+

**Is your feature request related to a problem? Please describe.**
High precision print using .25 nozzle, inconsistent 1st layer issue due to minor ~~PEI sheet~~ hot bed warping (thanks octoprint) on some corners

[image]

  
**Describe the solution you'd like**

I would like to see the 7x7 bed leveling included in the MINI firmware too :)

Refer to prusa3d/Prusa-Firmware#1239 where it was implemented on all the Prusa FDM line-up except the mini

It seems Prusa Slicer use G29 for bed leveling of the MINI, instead of G80.

No settings in regards of MBL in the settings menu of the printer

100jecki, 2021-12-02:
Up.

willbelr, 2022-01-27:
Please, this is badly needed. I spent so much time tweaking belt tension and live Z adjust, only to figure out the bed is badly warped. The auto bed leveling feature was a big selling point for which I chose the Mini, in practice this coud be easier really.

GFunkRailroad, 2023-03-07:
Please make this happen so I don't have to find custom firmware

Prusa-Support, 2023-04-26:
Thanks for reaching us.

Please allow me to close this issue as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1264.

For the time being - besides what already stated on https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1264#issuecomment-1365402581 - it may be useful to [correct the printer's geometry](https://help.prusa3d.com/article/squaring-your-mini_158518).
Optionally, re-do the [heated assembly finalization](https://help.prusa3d.com/guide/4-print-head-heatbed-assembly_201450#202585) while the heated is at about 100°C too. 

Michele Moramarco
Prusa Research

closed 2023-04-26 by Prusa-Support