prusa3d/Prusa-Firmware-Buddy #1520: [BUG] Watchdog reset

inferior-minion, 2021-07-29:
**Printer type** - [MINI] - Purchased 09/2019, received 03/2020. Used for thousands of hours already.

**Printer firmware version** - [e.g. 4.3.2 RC2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint**
  I print via USB flash drive - a new known good drive. The one that came with my printer was corrupt.

**Describe the bug**
  After an hour or so of printing the Mini stops mid-print and beeps loudly.

**How to reproduce**
  Start a print.

**Expected behavior**
  The Mini should finish the print.

**G-code**
  Please attach the G-code
[dump and gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/6903571/dump.and.gcode.zip)

**Crash dump file**
  See zip above.

**Video**
  I'll attach a picture from my phone shortly.

**Additional information**
  I've noticed that this happens with nearly every RC build on my specific printer but has almost never happened on a release fw build.

inferior-minion, 2021-07-29:
[image]

inferior-minion, 2021-07-29:
Clearer picture:

[image]

inferior-minion, 2021-07-30:
Update - After reverting to 4.3.1 and re-printing the exact same GCode, the print was successful on the first try.

JohnnyDeer, 2021-08-03:
Hi @inferior-minion, there is small chance that this is FW bug, most likely you really have broken thermistor cable. Would be good to observe if print stop on same spot again and if just before stop the bed temp drop down. You can print without filament with FS turned off to not waste your filament.

inferior-minion, 2021-08-05:
I haven't had this issue on 4.3.1 or other final release firmwares though...

Would you like me to update to 4.3.2 RC2 and then try to print the same gcode again without filament?

JohnnyDeer, 2021-08-09:
Yes, you can try it on 4.3.2 final, but it's unlikely that it happens only on RC. There was no changes touching watchdog. Most likely you really have broken thermistor cable triggering in some specific move.

inferior-minion, 2021-08-12:
I updated to 4.3.2 final and printed the same gcode with no trouble. I've also printed 10 or more newly sliced files with no issues.

I have noticed that sometimes when I have touched the Y axis cable while printing it has caused this (thermistor) issue. Should I request a replacement cable from support?

JohnnyDeer, 2021-08-23:
@inferior-minion If you want to request replacement you should have proof that your thermistor is broken. If you will experiencing this issue in future and will be reproducible please contact you Tech Support with video of the issue.
I'm closing this issue as "can't reproduce".

closed 2021-08-23 by JohnnyDeer