prusa3d/Prusa-Firmware-Buddy #1535: [BUG] Preheat Error

sht2017, 2021-08-07:
**Printer type** - [MINI]

**Printer firmware version** - [4.3.1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [None]

**USB drive or USB/Octoprint** - [No]

**Describe the bug**
When I'm painting the second layer, the printer would show PREHEAT ERROR (Check the print head heater & thermistor wiring for possible damage)

**How to reproduce**
Disable fans.

**Expected behavior**
Normally painting.
I've checked the support page (which is https://help.prusa3d.com/en/article/preheat-error-print-head-12202-mini_154797?errors=&fw=v4.3.1). It said, the message "Preheat error print head" is shown when the printer is unable to heat up the hotend more than 2 ºC for 20 seconds. Maybe do the temp test after disabling the fans in firmware?

JohnnyDeer, 2021-08-09:
Hi @sht2017, can you please try 4.3.2 final? Does it happen every time on same spot?

JohnnyDeer, 2021-09-02:
This issue is most likely caused by faulty heatbed thermistor cable and will be closed soon if autor will not respond.

closed 2021-09-06 by JohnnyDeer

Woelpi, 2022-10-14:
I‘m facing a similar issue. I run my prusa Mini in an enclosure with a sock on the heater block. So cooling of the heater block isn’t easy. 

With small parts in PETG and sequential printing, the heater isn’t cooled properly. The temperature change for the first layer doesn’t require any heating I suppose, because it cooles very slowly. When the heater block switches like this: 250C - 240C - 250C, I get a #12202 preheat error mid print on the 2nd-3rd layer on 5-10% of my parts on the bed. 

Checked all the wiring and wiggled an all the cables. All seems to be ok and the error sounds almost like this issue (low cooling, changes of different heating stages). 

It seems like there could be a hiccup in the safety algorithm with fast temperature changes. 

Firmware is the 4.4.0rc1.

Prusa-Support, 2022-10-14:
Hello @Woelpi, thank you for the report. On the original post fans were disabled, but how was it on your case? Hotend fan at full, regular speed and only print fan at low speed, or deactivated? It might be related with the safety algorithm from the description. I'll post everything to our devs so that we can take a more in-depth look. Thank you! 

Alessandro Pantaleo
Prusa Research

Woelpi, 2022-10-15:
Hello Alessandro,

the fans are running in accordance to the unmodified Prusament PETG profile. I didn’t modify any fan speeds. No print fan for the first 3 layers and than 30%-50% fan speed, as set in the standard profile.

When I print this small model I see that the temperature is all over the place. It drops as low as 233°C and then shoots to 246 °C for the first layer which is set at 240°C. Maybe it is all due to the sock insulating the heater block, which throws off your expected thermal characteristics for switching the temperature, as the heat flow to the outside air is significantly lower.