prusa3d/Prusa-Firmware-Buddy #1536: [BUG] 4.3.2 release has same build number as RC2

thsturgill, 2021-08-09:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - 4.3.2

**Original or Custom firmware** - [Original]

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive t**
  Please let us know if you print via USB flash drive or USB/Octoprint

 new firmware does not load over rc2 version.
**How to reproduce**
  copy new firmware to flash and reboot.

**Expected behavior**
  new firware would be flashed

JohnnyDeer, 2021-08-10:
Hi @thsturgill, it's possible to flash final over RC, you can flash even exact same FW over again. Please check if you have allowed flashing in Settings - FW Update.

Gordontgopher, 2021-08-10:
@JohnnyDeer I'm also seeing this issue and I do have updates set to 'always'. I have tried two USB sticks with the same result.

JohnnyDeer, 2021-08-11:
@Gordontgopher, so you have FW update to always, hit restart button, push knob for few seconds during restart and nothing happens?

Gordontgopher, 2021-08-11:
@JohnnyDeer yes that's right, I tried 4 or 5 times to try and eliminate user error, but same each time. I downloaded RC2 again and that flashed first time!

stixpjr, 2021-08-13:
I'm seeing this too, regardless of "FW Update" setting, "Always" or "On Restart" - 4.3.2-RC2 remains.

weust1979, 2021-08-13:
Same here. Can't update from RC2 to final.
Tried the options mentioned above.

Tsippaduida, 2021-08-14:
I finally got 4.3.2 final installed over RC2. Took umpteenth tries for two days before finally it loaded. I also removed all other firmware files from the USB-disk. FW-Update as "Always", and original USB-key in use. Previously it has always noticed when I insert the USB-stick and there's new firmware, but 4.3.2 final required reset + knob press and even then didn't react easily. The removal of other firmware versions from the USB-disk was made to help Mini to decide which one to load, but even that didn't seem to make any difference. Now finally when I tried yet another time, it worked (printer has been on for more than 12 hours since last re-flash attempt.

Gordontgopher, 2021-08-16:
After 2 or 3 attempts (with FW update set to 'restart') I got my mini to flash to 4.3.2 RC1, and successfully flashed to 4.3.2 from there.

JohnnyDeer, 2021-08-17:
Tested on three printers. I have flashed RC2 on MINI, download final to USB and RC2 leave there. FW update set to "On Restart" or "Always" (tested with both). Then I hit restart button, wait one secont to start printer booting, then click knob for one second and flashing dialog is shown, always, with and without factory reset.

Gordontgopher, 2021-08-17:
Appreciate there is not much you can do without being able to reproduce, is there anything we can do to supply debug info? would board version numbers or anything help?

stixpjr, 2021-08-17:
20th time lucky, I managed to flash mine. I'm thinking it's all about timing.

- push reset.
- push and hold selector-knob, but, like the instructions say, it **must** be released before the "verifying" screen appears. I think I've been holding it a few milliseconds too long.

thetable123, 2021-08-18:
Same issue here, had to sit and bump the rotary encoder during boot/POST before I got the option.

JohnnyDeer, 2021-08-23:
Honestly we do not consider this a bug. The final version compared to RC2 has no changes, so they have the same build number. This is the first time we have managed to create a good enough RC so that no further changes have to be made to it. 
There are now two options for users. The first is to leave printer with RC2 (it is really identical to the final version).
The second option is to load the final version as described above - give the bootloader a command to flash by clicking the knob while printer is booting.
Thank you all for understanding.

weust1979, 2021-08-23:
@JohnnyDeer Then I would like to suggest to mention in the changelog for Final that there are not changes since RC2 and that Final is essentially RC2.
I am fine now with running RC2 till the next version, but would have liked to know there were no changes since RC2 to Final.

JohnnyDeer, 2021-09-13:
@weust1979 Yes next time when this happen we will mention it in release notes.
So when we found out what's wrong I think this issue can be closed. Do you agree @thsturgill?

thsturgill, 2021-09-13:
Yes, especially since there is a new release.

closed 2021-09-13 by thsturgill