prusa3d/Prusa-Firmware-Buddy #1542: [BUG] Filament runout sensor still not working with octoprint

theHalo, 2021-08-15:
**Printer type** - [MINI+]

**Printer firmware version** - [4.3.2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint** - [Octoprint]

**Describe the bug**
I upgraded the firmware to 4.3.2 a few days ago. Now my filament ran out without anything happening (kept printing despite filament being past the sensor). Testet it twice, same result. 
When changing the filament the sensor works as expected. Restarting printer and octoprint didn’t help. 

**How to reproduce**
See above. 

**Expected behavior**
Automatically trigger local filament change when filament runs out during serial octoprint printing.

Danafra, 2021-08-16:
Agreed. It was working with the release candidate but the final release is broken.

d4rken, 2021-08-18:
Can confirm:
* Does not work: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.2
* Works: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.2-RC1

Octoprint Version 1.6.1

On a quick glance, I can't see though why it would suddenly stop working:
https://github.com/prusa3d/Prusa-Firmware-Buddy/compare/v4.3.2-RC1...v4.3.2
The additional commits don't touch any of the code that added the filament sensor support:
https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/f4338f3f40506b032acde284bf4248f1584c7ec1

Is the attached `.bbf` file maybe build of the wrong branch? Can't build and flash it myself due to signing :frowning:

rncfiles, 2021-08-18:
Also confirmed.  Reported to support chat.  

Important to note that the bug is more broad as sensor does not function with direct file printing (usb stick) or with serial printing.  Reverting to 4.3.1 confirms direct file printing works as expected.

mcdonaldAJCHOBBIES, 2021-09-17:
Also occurs in version 4.3.1+2759

Using USB connection

mcdonaldAJCHOBBIES, 2021-09-18:
v4.3.2-RC1 does not work for me

Using Printrun

elxris, 2021-11-06:
Bumping.

Not working with 4.3.3.

theHalo, 2021-11-06:
> Not working with 4.3.3.

Can confirm. Still not working reliably in 4.3.3
Had it working twice and failing four or more times.

TerTerro, 2021-11-11:
Just had Filament Sensor trigger and octoprint did not get message.

JohnnyDeer, 2021-11-22:
Dear @theHalo, we tested your issue and it seems to only affect older versions of Octoprint. Could you please try upgrading to Octoprint 1.7.2 and report back? Thank you.

your-friend-alice, 2021-12-02:
I just tested on Octoprint `1.7.2` + Firmware `4.3.3`, seems to work. my only non-stock plugin is `M73 ETA Override (1.0.4)`.

theHalo, 2021-12-13:
I wasn’t sure when I did Update to current stable Octoprint. So I tried a few times cutting the filament. It worked. A few days later I actually ran out of filament and the Mini failed to do anything. Kept printing midair. 
It failed a second time a bit later and then I tried manually and it failed again. I even inserted filament into the sensor and pulled out again multiple times. No reaction. 

The sensor still works **_every_** time during filament changes but it seems I can’t get it working reliably during Octoprint printing. 

Octoprint settings? Plugins? Anything?

My active plugins:
Bed Visualizer, BetterHeaterTimeout, Custom Control Editor, Dashboard, Detailed Progress, DisplayLayerProgress Plugin, Draggable Files, MultiCam, Octolapse, OctoPod Plugin, PrettyGCode, PrintJobHistory, PrintTimeGenius Plugin, Resource Monitor, Slicer Thumbnails, Tab Order, TouchUI

theHalo, 2021-12-19:
**Reproducibility at last!**

1. Start test print from Octoprint. 
2. Cut filament: filament sensor works, detects runout, initiates filament change, resumes print (tested multiple times). 
3. **Enter „Tune“ Menu on the Mini and change a setting (tested with Nozzle Temp, Bed Temp, Filament Change, Speed, Fan, Flow)**
4. **Do NOT exit „Tune“ Menu.** 
5. Cut filament: filament sensor **_fails_** to detect runout, nothing happens when the filament end is past the sensor (tested multiple times). 
6. To resume normal functionality of the filament sensor simply exit the „Tune“ Menu. 

Rebooting Octoprint and disconnecting/reconnecting also re-establishes normal filament sensor behavior. 

Can anybody check to verify this?

coaxial, 2022-02-23:
Octoprint 1.7.3 here with 4.3.3 and 4.3.4. If I leave the "messages" menu open (since Prusa still hasn't implemented showing messages on the main serial screen), the filament sensor does nothing. 

If I close it and exit the tube menu, then it works.

doodlebro, 2022-03-12:
I can confirm that 4.3.4 does not have consistent filament sensor behavior when controlled by octoprint.

Restarting octoprint, the printer, and leaving the printer menu alone is the only solution right now. I expect the filament sensor to work no matter what has been done in the serial printing menu, nor what page of the menus I am on.

ScubaDamX, 2022-05-01:
I had the same issue in 4.3.4 I had the Live Z adjust menu open and the filament sensor didn't work so lost my entire print. I tested the sensor from USB and worked perfectly. I wish I had known you HAVE to have the Octoprint logo on the Prusa Mini screen for the filament sensor to work!

closed 2024-06-17 by github-actions[bot]