prusa3d/Prusa-Firmware-Buddy #1543: [BUG] FW 4.3.2 (8/9/21) Breaks z-home & crash detection

DevinCharles, 2021-08-15:
**Printer type** - MINI

**Printer firmware version** - 4.3.2 (8/9/21)

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
  USB Drive

**Describe the bug**
  Just upgraded to 4.3.2 from 4.3.2 RC, now on z-home my nozzle is burried into my build plate and it keeps driving, until I push the reset button. Similarly, moving to the highest extent when the printer is powered off with the print head higher than 0, the print head hits the upper stop and continues to drive.

**How to reproduce**
  Flash 4.3.2, attempt to home.

**Expected behavior**
  Developers test their "release" level firmware to make sure it doesn't ruin hardware, Z stop works correctly.

**G-code**
  N/A

**Crash dump file**
  N/A

**Video**
  N/A

JohnnyDeer, 2021-08-17:
Hi @DevinCharles, I can assure you that all FW versions are heavily tested before release.
This sound like your MINDA is broken. Please try if your MINDA works - https://help.prusa3d.com/en/article/m-i-n-d-a-superpinda-sensor-testing-mini-mini_134021
If you find out that your sensor is broken please contact our Tech Support. 
BTW MINI don´t have crash detection.

Prusa-Support, 2022-05-22:
Hello @DevinCharles! Can you please update me on the issue, was it resolved in the end through our Technical Support? In case of anything to still review, do not hesitate to let me know via email info@prusa3d.com mentioning me ("attn: Alessandro"). Closing the issue for now as it did look not firmware related, but more related with the component itself.

Alessandro Pantaleo
Prusa Research

closed 2022-05-22 by Prusa-Support