prusa3d/Prusa-Firmware-Buddy #1575: [BUG] Wrong text / Traduction

Mryck, 2021-09-07:
**Printer type** - [MINI, MINI+]

**Printer firmware version** - [4.3.3 RC]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint**
 USB Printing

**Describe the bug**
 Tinkering with the new version and menu came across a strange text in the hidden Hardware menu.
(Thanks for this nice FW update)

**How to reproduce**
Go to Hidden menu, *French Translation*

**Expected behavior**
Something clear at least in English.

[image]

JohnnyDeer, 2021-09-09:
Hi @Mryck, thank you for reporting.

Mryck, 2021-10-15:
Hi @JohnnyDeer thanks it's fixed on official release

closed 2021-10-15 by Mryck