prusa3d/Prusa-Firmware-Buddy #1576: Blue screen with octoprint connected

Madmatt06, 2021-09-07:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer Type**
Prusa mini +

**Printer firmware version**
4.3.2

**Original or Custom firmware**
Original

**Optional upgrades**
Filament Runout Sensor

**USB drive or USB/Octoprint**
Octoprint

**Describe the bug**
When the printer starts purging the filament and octoprint connects with micro usb after confirming correct color printer goes back to start up screen then displays blue screen with error.

**How to reproduce**
Start purging filament and tell octoprint to connect via webpage while purging.

**Expected behavior**
Printer should go to octoprint screen when correct color is confirmed.

**G-code**
There was no G-code ran at the time

**Crash dump file**
[dump.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/7124679/dump.bin.zip)
I apologize if I didn't upload it correctly. This is my first time creating an issue.

**Video**
I only have an image of the screen during the error.
[image]

kcwu, 2021-09-08:
FYI, I have very similar issue #1578.

JohnnyDeer, 2021-09-20:
Hi @Madmatt06, we are unable to reproduce this issue by following step described here. Can you please upload video with all steps you did before causing BSOD? Thank you

kcwu, 2021-09-20:
@JohnnyDeer , could you reproduce with steps in #1578?

RobeeeJay, 2021-09-21:
I have experienced this a few times, it seems quite reproducible for me. Here is how it happens:

1. Power on the Mini and OctoPi at the same time
2. Before OctoPrint has connected to the Mini, enter the filament menu and Change filament
3. OctoPrint will say it is connected but the Mini's UI will not
4. When the filament has been changed and you say yes to the colour change is good, the Mini will instantly reboot and once the firmware has loaded it will then throw a BSOD

Madmatt06, 2021-09-21:
Here is the video for it.
https://youtu.be/L8KiI6BOAR4

tumtumtum, 2022-04-13:
I have this exact problem as well.

Prusa-Support, 2022-06-26:
Hello! Sorry for the late response on this - I noticed it seems to be the same as #1683. Can you please attempt disconnecting the Extruder Motor temporarily, and then "command the unload filament"? Would it still do the blue screen that way? We suspect a possible under-voltage measured on motor possibly causing this. 

Alessandro Pantaleo
Prusa Research

Madmatt06, 2022-07-06:
@Prusa-Support I reproduced the error with the extruder connected to confirm its still there and it was. I then disconnected the extruder and tried to reproduce the error and it didn't occur. After reconnecting the extruder I am no longer Able to reproduce it.

Prusa-Support, 2022-07-19:
Hello @Madmatt06, thank you for your feedback. Good to hear it seems to have been solved, so it could have been the "under-voltage" on motor but still we are not 100% sure. Let's of course keep an eye on the situation in case it might occur again. 

For anyone with the same or similar issue, feel free to add more comments and feedback, even after the mentioned test. It will help us better understanding the possible cause - of course if it occurs again, we can absolutely further review. 

Alessandro Pantaleo
Prusa Research

AronsPC, 2022-07-25:
I have this same issue. It's happened to me four times now on one of my Minis. Firmware 4.3.4. I have re-flashed as well after one of the errors and it just happened again on me. Each time it was after a filament unload it would reset the machine and then bluescreen
[image]
.

Prusa-Support, 2022-07-27:
Hello @Aron3DP! Thank you for the report and sorry for the issue. Have you attempted the workaround I posted on June 26, on a previous comment? The one with disconnecting the motor? Would it still do the blue screen during that test? 

Alessandro Pantaleo
Prusa Research

Aron3DP, 2022-07-28:
@Prusa-Support I did not. I missed that line just a few above. I will see about doing so and give it a try.

bmorgenthaler, 2022-11-21:
I just had a similar BSOD but my scenario was a bit different:

1. MINI on 4.4.0-RC1 running, print uploaded and started through PrusaConnect
2. Octoprint running but not connected (USB physically disconnected)
3. USB physically connected
4. Octoprint system rebooted
5. Because of Autoconnect enabled, Octoprint tried to connect to the serial
6. BSOD on displayTask

Unfortunately I did not get a photo of the exact error message, will try again.

Prusa-Support, 2022-12-09:
Hello! Thank you for the report, all noted. Let me know if it might occur again, systematically, or even with the 4.4.0 final release. 

Alessandro Pantaleo
Prusa Research

tumtumtum, 2023-03-24:
> Hello @Madmatt06, thank you for your feedback. Good to hear it seems to have been solved, so it could have been the "under-voltage" on motor but still we are not 100% sure. Let's of course keep an [...]

Hi Alessandro, can you explain more about the under voltage on the extruder motor? Is there a bug in the firmware related to the motor not getting enough voltage? Is this fixable with a firmware update? What would cause an under voltage (loose connections etc)?

closed 2024-06-17 by github-actions[bot]